ArduPilot/MissionPlanner #2413: MAVFTP usability issues when combined with MP

rmackay9, 2020-06-20:
I am unsure if these are Mission Planner issues or flight code so I'm documenting these issue but here and in the AP flight code issues list:

All of these issues can be reproduced by doing the following:

- Open Mission Planner
- Open the Simulation page
- Ensure the "wipe" checkbox is checked
- Press the Copter icon at the bottom
- Open Config, MAVFtp

When trying to upload a Lua script to a copter in MP's simulator I've run into these problems:

- Click, "@ROMFS", then click on the unnamed (top directory?), 5 second delay before the directory listing is displayed.  5sec is not very long but it is very noticeably longer than if I click on one of the sub directories (i.e. logs) and then switch back to the top directory
[image]
- Attempting to create a directory with the same name as an existing directory causes MP to lock-up for 30 seconds.  The directory is not created and no error message is displayed
- Attempting to delete an empty directory causes MP to lock-up for 30 seconds.  A "Failed to delete file" message is displayed and the directory is not deleted

Hwurzburg, 2020-06-25:
another issue being found repeatedly is that on 56K Telem links the " looking for Mavftp param" can take 30seconds or more before reverting to normal param fetching...it can be circumvented by closing that dialog....but I dont think it should be necessary...

meee1, 2020-06-25:
@Hwurzburg  realy need to test this against the current beta, ive addressed all the already

rmackay9, 2020-06-25:
I've tested this and all the issues seem resolved thanks!

As a side note I was able to create a @ROMFS directory.  It didn't seem to do any harm though.

closed 2020-06-25 by rmackay9

Hwurzburg, 2020-06-26:
thanks...will pass along to those having issues