ArduPilot/MissionPlanner #2420: Plan: map jumps to 0,0 when pressing Home Location

Hasi123, 2020-06-25:
#### Issue details
On the planning tab if you press Home Location, the map jumps to coordinates 0,0 instead of the newly set home location.

#### Version
The issue appeared with version 1.3.71

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2020-06-25:
pressing home location sets home the the current vehicle location. did you have anything connected?

Hasi123, 2020-06-26:
Yes, the copter is connected and the home location also gets set succesfully, but the map jumps to coordinates 0,0 (which is somewhere in the middle of the ocean off the western african coast) instead of the home location like it used to.

mflane, 2020-06-29:
Can confirm issue on 1.3.72 - Copter 4.0.3

Hasi123, 2020-10-28:
Issue is still present in 1.3.74

Hasi123, 2021-02-08:
Any fix for this? It is quite annoying...