ArduPilot/MissionPlanner #2422: MAVFtp not working/available for components anymore

olliw42, 2020-06-30:
#### Issue details
MAVFtp not working/available for components.

MAVLink components may also support the MAVFtp protocol, specifically e.g. a camera, or a gimbal. In earlier MP versions it was possible to access their files, this seems to be not possible anymore. Not found in the menu, nor cntrl-F nor cntrl-X.

#### Version
1.3.72 build 1.3.7463.24854

#### Platform
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
irrelevant

#### Hardware type
irrelevant

#### Logs
irrelevant

meee1, 2020-07-01:
mavftp has moved to the main menus now.
but it is behind a support bit
https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L4969
which is inside
mavlink_msg_autopilot_version_send

olliw42, 2020-07-01:
the camera and gimbal I'm talking about are sending this bit set in their autopilot_version messages (they btw do so since they support MAVFtp)

meee1, 2020-07-01:
ok, then AFAIK it should appear as mavftp in Config screen

olliw42, 2020-07-01:
it doesn't. it does for the autopilot, but not for the gimbal and camera

meee1, 2020-07-01:
even if you pick the gimbal from the menu on the top right?

olliw42, 2020-07-01:
yes, sure. that's the only way to get the gimbals parameters etc, isn't it

meee1, 2020-07-01:
correct. the config screens works based on the current "active" selected mavlink device.

olliw42, 2020-07-01:
yeah, so my answer was correct, I see the mavftp option when I chose the autopilot, but not then I chose the gimbal or camera

olliw42, 2020-07-01:
is it possible that something has changed as regards the commandlong and it's ack handling in MP?

meee1, 2020-07-01:
no that hasn't changed. what i think might be happening is MP is not requesting the "autopilot_version" message for the component

meee1, 2020-07-01:
if it was the only connected device i think it would work, but if there or more than one endpoint i think it never requests it from the other component

meee1, 2020-07-01:
i do have a check, but i asume it fails the armed check
                                    // poll for version if we dont have it - every mav every port
                                    if (!port.giveComport && !MAV.cs.armed && (DateTime.Now.Second % 20) == 0 && MAV.cs.version < new Version(0, 1))
                                        port.getVersion(MAV.sysid, MAV.compid, false);

olliw42, 2020-07-01:
this would be "interesting" since it used to be important for a component to support autopilot_version since otherwise MP would not handle it friendly
I guess I should be able to test if MP requests this message

olliw42, 2020-07-01:
ah ... these components won't be armed at startup, and they may not be so for a long while (depending on conditions)

meee1, 2020-07-01:
yep, i think the fix is for me to remove the armed condition, then it will just work

olliw42, 2020-07-01:
I think for an autopilot it probably makes sense, but probably not so for components

I guess an alternative could be to forbid ALL mavftp when teh autopilot is armed, but allow all otherwise, i.e., to make it dependent only on the autopilot armed but not the component armed

meee1, 2020-07-01:
ive just pushed a change to MP beta, that should work for you. will take 5-10 mins to build in CI

olliw42, 2020-07-01:
I will test tomorrow :)
thx so far

olliw42, 2020-07-01:
tested ... and ... WORKS! great. Thx.

FYI, when I download a file I get the error
[image]
I'm using "download" not "download burst", and this is for any file, also the files on the ardupilot (so I assume it's an MP thing and not camera or gimbal thing).

closed 2023-11-14 by olliw42