ArduPilot/MissionPlanner #2424: Shortcut to read or write waypoints

expntly, 2020-07-01:
#### Issue details
I couldn't find a shortcut (hotkey) to read or write waypoints.  MP is really great but so many features are hidden, sorry if I missed this one but I've looked everywhere.  Could this please be added?  Thanks

#### Version
1.3.72

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2020-07-01:
Currently there is non to read/write wps.
there are to open a wp file, and save a wp file
control-z - undo
control-o - open wp file
control-s - save wp file

meee1, 2020-07-01:
https://github.com/ArduPilot/MissionPlanner/blob/3bca203ad7ee4ee0cd409431f42a8bcc467b1ffa/GCSViews/FlightPlanner.cs#L331

expntly, 2020-07-05:
Is this something one can add via a py script or plugin?

expntly, 2020-07-07:
I did some more digging and found that https://github.com/ArduPilot/MissionPlanner/blob/master/plugins/Shortcuts/Plugin.cs, so I guess one can write their own C# plugin to add shortcuts.

I looked into whether I could do this with python but I can't find that `MainV2` or `MainV2.instance` object exposed: https://github.com/ArduPilot/MissionPlanner/blob/9d8cef76f1b14fd2627afd2a01ae1081f134ad62/Script.cs#L45

Is there a way to add a `ProcessCmdKeyCallback` in a python script?

expntly, 2020-07-27:
ping @meee1 

If that's not possible in Python this issue can be closed.

meee1, 2020-07-27:
technically yes you can, i  have not tested it however

meee1, 2020-07-27:
what key combo did you have in mine? i can add it to beta MP

expntly, 2022-04-30:
@meee1 Any key combo will do (ctrl+shift+w?), thanks!

meee1, 2022-05-04:
added

closed 2022-05-04 by meee1

sinegole, 2022-07-18:
Hi Michael, The ctrl+shift+W is shortcut for "Write fast", could you please add shortcut for standard "Write".