ArduPilot/MissionPlanner #2426: MAV.SubscribeToPacketType(STATUSTEXT) not showing all messages

expntly, 2020-07-07:
#### Issue details

I noticed `MAV.SubscribeToPacketType(STATUSTEXT)` seems to filter some packets depending on who's sending it.  If it is the flight controller emitting messages, the callback will get called.  But if it's any other MAVLink nodes, it won't.  I know MP is receiving these STATUSTEXT messages because `MAV.readPacket()` returns these.  Because `readPacket` has its own problems racing with the main thread, I'd like to find a way to make `SubscribeToPacketType` work for STATUSTEXT messages.

I use the following script.

```python
import clr
import MissionPlanner
clr.AddReference("MAVLink")
from System import Func
import MAVLink

def OtherMethod(message):
    print "got HB";
    return True

def MyMethod(message):
    print("STATUSTEXT from MAV.SubscribeToPacketType")
    print(message.data.text)
    return True

sub = MAV.SubscribeToPacketType(MAVLink.MAVLINK_MSG_ID.HEARTBEAT.value__, Func[MAVLink.MAVLinkMessage, bool] (OtherMethod))
sub2 = MAV.SubscribeToPacketType(MAVLink.MAVLINK_MSG_ID.STATUSTEXT, Func[MAVLink.MAVLinkMessage,
                                 bool] (MyMethod))
sub2 = MAV.SubscribeToPacketType(MAVLink.MAVLINK_MSG_ID.STATUSTEXT_LONG, Func[MAVLink.MAVLinkMessage,
                                 bool] (MyMethod))

import time
while True:
    p = MAV.readPacket()
    if p.msgtypename == "STATUSTEXT":
        print("STATUSTEXT from MAV.readPacket %s" % p.data.text)
    time.sleep(0.01)

```

To test I just start a sitl instance and open one connection with MP, and another one with mavproxy.py and do a `module load msg`.  If I type `msg hello`, MP will only show it received it via readPacket and not the subscribed callback.

```
STATUSTEXT from MAV.readPacket Array[Byte]((104, 101, 108, 108, 111, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))
```

But if I do a preflight calibration to make the FC "talk", I see that both readPacket and the subscribed callback see the STATUSTEXT messages.

```
STATUSTEXT from MAV.SubscribeToPacketType
Array[Byte]((66, 97, 114, 111, 109, 101, 116, 101, 114, 32, 99, 97, 108, 105, 98, 114, 97, 116, 105, 111, 110, 32, 99, 111, 109, 112, 108, 101, 116, 101, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))
STATUSTEXT from MAV.readPacket Array[Byte]((66, 97, 114, 111, 109, 101, 116, 101, 114, 32, 99, 97, 108, 105, 98, 114, 97, 116, 105, 111, 110, 32, 99, 111, 109, 112, 108, 101, 116, 101, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))
STATUSTEXT from MAV.SubscribeToPacketType
Array[Byte]((85, 112, 100, 97, 116, 105, 110, 103, 32, 98, 97, 114, 111, 109, 101, 116, 101, 114, 32, 99, 97, 108, 105, 98, 114, 97, 116, 105, 111, 110, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))
STATUSTEXT from MAV.SubscribeToPacketType
Array[Byte]((65, 105, 114, 115, 112, 101, 101, 100, 32, 99, 97, 108, 105, 98, 114, 97, 116, 105, 111, 110, 32, 115, 116, 97, 114, 116, 101, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))
STATUSTEXT from MAV.SubscribeToPacketType
Array[Byte]((65, 105, 114, 115, 112, 101, 101, 100, 91, 48, 93, 32, 115, 101, 110, 115, 111, 114, 32, 99, 97, 108, 105, 98, 114, 97, 116, 101, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))
STATUSTEXT from MAV.readPacket Array[Byte]((65, 105, 114, 115, 112, 101, 101, 100, 91, 48, 93, 32, 115, 101, 110, 115, 111, 114, 32, 99, 97, 108, 105, 98, 114, 97, 116, 101, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))
```

So somehow `SubscribeToPacketType` only sees messages from the FC, not the other MAVLink nodes.  Am I missing something?

#### Version
1.3.70

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2020-07-07:
i dont see ".value__" on the enums

meee1, 2020-07-07:
After rereading the problem it's because it's from a different component that the current selected

meee1, 2020-07-07:
https://github.com/ArduPilot/MissionPlanner/blob/master/plugins/example.cs
That's c# code that listens on every packet across the link. Ie all components just need to translate to python

meee1, 2020-07-07:
here is a working example

import clr
import MissionPlanner
clr.AddReference("MAVLink")
from System import Func, Action
import MAVLink

def OtherMethod(message):
    print "got HB";
    return True

def MyMethod(message):
    print("STATUSTEXT from MAV.SubscribeToPacketType")
    print(message.data.text)
    return True

def MyPacketHandler(o, message):
    try:
        if message.msgid == MAVLink.MAVLINK_MSG_ID.STATUSTEXT.value__:
            print "STATUSTEXT from MyPacketHandler " + str(message.sysid) + " " + str(message.compid)
            print dir(message)
            print(bytes(message.data.text))
    except Exception as inst:
        print inst

sub = MAV.SubscribeToPacketType(MAVLink.MAVLINK_MSG_ID.HEARTBEAT.value__, Func[MAVLink.MAVLinkMessage, bool] (OtherMethod))
sub2 = MAV.SubscribeToPacketType(MAVLink.MAVLINK_MSG_ID.STATUSTEXT, Func[MAVLink.MAVLinkMessage,
                                 bool] (MyMethod))
sub3 = MAV.SubscribeToPacketType(MAVLink.MAVLINK_MSG_ID.STATUSTEXT_LONG, Func[MAVLink.MAVLinkMessage,
                                 bool] (MyMethod))

MAV.OnPacketReceived += MyPacketHandler

import time
while True:
    time.sleep(1)

expntly, 2020-07-07:
That method is the way to go, it works great!  Thanks for the fast turnaround time!!

closed 2020-07-07 by expntly