ArduPilot/MissionPlanner #2430: 3D Map broken after version 1.3.68

expntly, 2020-07-14:
#### Issue details
3D Map works fine with 1.3.68.

With 1.3.69 however, the ground is very blurry, and with 1.3.71 it basically stopped working: the ground is either all black or blue.  I tried adjusting the two numbers in the top left corner (no idea what they are) and enable/disable fog and "lock to mav" but that doesn't help.

#### Version
See above.

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2020-07-14:
at a guess this is because of a graphics card issue. ie its too old. what graphics card do you have?

meee1, 2020-07-14:
[image]

expntly, 2020-07-15:
This is actually a brand new laptop with an Intel HD Graphics 620.  But it also does the same on other computers we've tried.  1.3.68 works everywhere, but not any of the later releases on our side.  Anyway, I updated the driver from 9/25/2019 to 6/1/2020 and that didn't help.  Here's what I see:

[image]

Has anything changed specifically after 1.3.68 and related to this 3D view?

meee1, 2020-07-16:
can you try the latest beta MP?

expntly, 2020-07-17:
Exact same behavior with the latest I just downloaded `Mission Planner 1.3.72 build 1.3.7463.24854`.

Could it be related to using the zip + extract instead of installing the msi?  I don't want to overwrite my working 1.3.68 system install, so I'm just downloading zip archives and executing locally from the extracted folder.

meee1, 2020-07-17:
You need to use beta. Goto help and click update beta

meee1, 2020-07-20:
ive managed to replicate this, and seems that you cant load a texture on a second shared context with intel cards. nvidia card works fine.  just bad drivers from intel......

expntly, 2020-07-21:
Does that mean MissionPlanner-latest has the fix?  I tried but I'm still seeing the same thing.

meee1, 2020-07-21:
no.
latest is the last stable version
you need to use the update beta button in the help screen

expntly, 2020-07-21:
Got it thanks.

It seems better thanks, although there are some strange horizontal artifact "lines".

[image]

Any idea what this is?

Also, what are the two numbers on the top left corner?

meee1, 2020-07-21:
the two numbers are the image zoom levels.  ie from zoom 12 to 20 are rendered. fog is not working atm. and lock to mav you can untick and fly around the map using ASDW pan QE alt RF
the line is the mouse pointer location

expntly, 2020-07-21:
I meant the **horizontal lines**.

meee1, 2020-07-22:
agh, thats overlapping textures. ive fixed that already/ bad blending options
[image]