ArduPilot/MissionPlanner #2433: ADS-B object do not update velocity.

mateusz-spychala, 2020-07-17:
#### Issue details
When ADS-B receiver is connected to Pixhawk. ADS-B object do not update velocity in MissionPlanner
visualization. 

#### Version
_What version was the issue encountered with_
1.3.72
#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Solution

I have checked code and one line is missing:
File: MainV2.cs
Function: adsb_UpdatePlanePosition:1190

                if (MainV2.instance.adsbPlanes.ContainsKey(id))
                {
                    // update existing
                    ((adsb.PointLatLngAltHdg)instance.adsbPlanes[id]).Lat = adsb.Lat;
                    ((adsb.PointLatLngAltHdg)instance.adsbPlanes[id]).Lng = adsb.Lng;
                    ((adsb.PointLatLngAltHdg)instance.adsbPlanes[id]).Alt = adsb.Alt;
                    ((adsb.PointLatLngAltHdg)instance.adsbPlanes[id]).Heading = adsb.Heading;
                    ((adsb.PointLatLngAltHdg)instance.adsbPlanes[id]).Time = DateTime.Now;
                    ((adsb.PointLatLngAltHdg)instance.adsbPlanes[id]).CallSign = adsb.CallSign;
                    ((adsb.PointLatLngAltHdg)instance.adsbPlanes[id]).Squawk = adsb.Squawk;
                    ((adsb.PointLatLngAltHdg)instance.adsbPlanes[id]).Raw = adsb.Raw;
**>>>>>> HERE IS MISSING UPDATE OF VELOCITY/SPEED <<<<<<<<<<<<**
                }

magicrub, 2020-08-19:
@meee1 ping!

mateusz-spychala, 2020-08-20:
Seems solved. Thank you.

closed 2020-08-20 by mateusz-spychala