ArduPilot/MissionPlanner #2436: MJPEG unstabilities

lebarsfa, 2020-07-24:
#### Issue details
With some MJPEG servers, Mission Planner on Ubuntu only shows a first image and after 10 s reconnects and do the same. This appears to be caused by the loop https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/CaptureMJPEG.cs#L149-L154 where I guess that a `Read()` operation of 0 bytes happens and might be interpreted by the server as a communication error and Mission Planner is then probably disconnected. This seems to be solved by changing a little the loop with something like : 
```csharp
                            int jpeglength = length;
                            while (offset < jpeglength)
                            {
                                len = br.Read(buf1, offset, length);
                                if (len <= 0) break;
                                offset += len;
                                length -= len;
                            }
```

#### Version
1.3.72

#### OS
Ubuntu 18.04 with Mono 6.10.0.104

lebarsfa, 2020-07-24:
#2357 was also related to MJPEG and I remember I got problems with MJPEG for some versions of Mission Planner but with the current version I did not see issues with the mentioned stream (https://webcam1.lpl.org/axis-cgi/mjpg/video.cgi) with Windows 10 v1909. However, for Ubuntu 18.04 with Mono 6.10.0.104, I got 
```csharp
ERROR MissionPlanner.Utilities.CaptureMJPEG - System.Net.WebException: Error: TrustFailure (Authentication failed, see inner exception.) ---> System.Security.Authentication.AuthenticationException: Authentication failed, see inner exception. ---> Mono.Btls.MonoBtlsException: Ssl error:1000007d:SSL routines:OPENSSL_internal:CERTIFICATE_VERIFY_FAILED
  at /build/mono-6.10.0.104/external/boringssl/ssl/handshake_client.c:1132
  at Mono.Btls.MonoBtlsContext.ProcessHandshake () [0x00048] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at Mono.Net.Security.MobileAuthenticatedStream.ProcessHandshake (Mono.Net.Security.AsyncOperationStatus status, System.Boolean renegotiate) [0x000da] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at (wrapper remoting-invoke-with-check) Mono.Net.Security.MobileAuthenticatedStream.ProcessHandshake(Mono.Net.Security.AsyncOperationStatus,bool)
  at Mono.Net.Security.AsyncHandshakeRequest.Run (Mono.Net.Security.AsyncOperationStatus status) [0x00006] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at Mono.Net.Security.AsyncProtocolRequest.ProcessOperation (System.Threading.CancellationToken cancellationToken) [0x000fc] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
   --- End of inner exception stack trace ---
  at Mono.Net.Security.MobileAuthenticatedStream.ProcessAuthentication (System.Boolean runSynchronously, Mono.Net.Security.MonoSslAuthenticationOptions options, System.Threading.CancellationToken cancellationToken) [0x00262] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at Mono.Net.Security.MonoTlsStream.CreateStream (System.Net.WebConnectionTunnel tunnel, System.Threading.CancellationToken cancellationToken) [0x0016a] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at System.Net.WebConnection.CreateStream (System.Net.WebOperation operation, System.Boolean reused, System.Threading.CancellationToken cancellationToken) [0x001ba] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
   --- End of inner exception stack trace ---
  at System.Net.WebConnection.CreateStream (System.Net.WebOperation operation, System.Boolean reused, System.Threading.CancellationToken cancellationToken) [0x0021a] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at System.Net.WebConnection.InitConnection (System.Net.WebOperation operation, System.Threading.CancellationToken cancellationToken) [0x00141] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at System.Net.WebOperation.Run () [0x0009a] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at System.Net.WebCompletionSource`1[T].WaitForCompletion () [0x00094] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at System.Net.HttpWebRequest.RunWithTimeoutWorker[T] (System.Threading.Tasks.Task`1[TResult] workerTask, System.Int32 timeout, System.Action abort, System.Func`1[TResult] aborted, System.Threading.CancellationTokenSource cts) [0x000f8] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at System.Net.HttpWebRequest.GetResponse () [0x00016] in <4e15bbae9d7043d8afd6cfd50bd9bd5a>:0 
  at MissionPlanner.Utilities.CaptureMJPEG.getUrl () [0x00043] in <1fb30d85febe4713b668fba9356951a6>:0 
```
I am not sure whether it is more a problem in Mono, missing certificates in my Ubuntu or another thing but from [this suggestion](https://stackoverflow.com/questions/54431583/is-this-the-correct-solve-for-certificate-verify-failed-exception), adding 
```csharp
ServicePointManager.ServerCertificateValidationCallback = (sender, certificate, chain, sslPolicy) => true;
```
before https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/CaptureMJPEG.cs#L87 appears to solve the problem for me. I do not know if this could have side effects though...

meee1, 2020-07-27:
Mono maintains its own CA list so need to install the CA for that server into mono

meee1, 2020-07-27:
As for the short read, a read return of 0 is meant to indicate end of stream. So mono is doing something wired here. I'll need to do some more testing

meee1, 2020-07-27:
[image]

meee1, 2020-07-27:
[image]
Looks like I might need to 0 check as well depending on the stream type

lebarsfa, 2020-07-27:
About the `Read()`, I meant it is when we ask explicitely to read 0 bytes in the parameters of the `Read()` function that the servers might want to disconnect the client, otherwise a returned value of 0 bytes probably only happens when the read timeout is reached, which looks OK to me. Therefore, the fix could be to always avoid asking to read 0 bytes (unless there was a specific reason to do so?).
Also, I should mention that this problem does not happen specifically with the stream https://webcam1.lpl.org/axis-cgi/mjpg/video.cgi but with another server that I have.

meee1, 2020-07-28:
regarding the cert
some info here
https://www.mono-project.com/docs/faq/security/#secure-socket-layer-ssl--transport-layer-security-tls

and im  looking into the read 0 issue

closed 2020-07-28 by meee1

lebarsfa, 2020-07-28:
Thank you, I confirm my MJPEG server works well with the latest beta on Ubuntu 18.04, 20.04 and Windows.
However, I could not get https://webcam1.lpl.org/axis-cgi/mjpg/video.cgi to work but it seems not much a Mission Planner-specific issue since it is related to its certificates validation with Mono (curl is also not happy with them) and anyway it was just an example of debug URL, others proposed in #2357 (e.g. http://bob.novartisinfo.com/axis-cgi/mjpg/video.cgi) work well.

meee1, 2020-07-28:
yeah, that url needs a cert installed, im sure if you fix curl the mono fix would most likely be free
[image]

meee1, 2020-07-28:
the certs
http://secure.globalsign.com/cacert/root-r1.crt
https://secure.globalsign.com/cacert/gsdomainvalsha2g2r1.crt

lebarsfa, 2020-07-28:
Thank you again, but I still could not get to work https://webcam1.lpl.org/axis-cgi/mjpg/video.cgi. I tried things like 
```bash
sudo certmgr -ssl https://valid.r1.roots.globalsign.com/
sudo certmgr -ssl https://webcam1.lpl.org/axis-cgi/mjpg/video.cgi
wget http://secure.globalsign.com/cacert/root-r1.crt
wget https://secure.globalsign.com/cacert/gsdomainvalsha2g2r1.crt
certmgr -add -c Trust ./root-r1.crt
certmgr -add -c Trust ./gsdomainvalsha2g2r1.crt
... (with sudo and without, -c CA, -c My, -c AddressBook, and other things...)
sudo mozroots --import --quiet
sudo mozroots --sync
```
I noticed on another Ubuntu virtual machine that Firefox also was not happy with that website (message `Warning: Potential Security Risk Ahead`, `Error code: SEC_ERROR_UNKNOWN_ISSUER`, and we need to `Accept the Risk and Continue`) so I guess there is a configuration problem on their side which probably requires to add an exception to accept it like probably Firefox let us do, but it looks quite outside the scope of Mission Planner.

meee1, 2020-07-29:
cd /usr/local/share/ca-certificates
sudo wget http://secure.globalsign.com/cacert/root-r1.crt
sudo wget https://secure.globalsign.com/cacert/gsdomainvalsha2g2r1.crt
sudo openssl x509 -in root-r1.crt -inform der -out rootr1b64.crt
sudo openssl x509 -in gsdomainvalsha2g2r1.crt -inform der -out gsdomainvalsha2g2r1b64.crt
sudo update-ca-certificates

csharp -e 'new System.Net.WebClient ().DownloadString ("https://webcam1.lpl.org/axis-cgi/mjpg/video.cgi")'

lebarsfa, 2020-07-29:
Perfect, it works!
Many thanks again!

commit daaebfb9 (2020-07-28): CaptureMJPEG: ubuntu fix #2436