ArduPilot/MissionPlanner #2437: Gstreamer compatibility on Ubuntu

lebarsfa, 2020-07-25:
#### Issue details
If I run `gst-launch-1.0 -e -v videotestsrc  do-timestamp=true ! video/x-raw, format=NV12, width=320, height=240, framerate=25/1 ! videoconvert ! x264enc ! h264parse ! queue ! rtph264pay config-interval=10 pt=96 ! udpsink host=127.0.0.1 port=5600` and then launch Mission Planner on Windows, it will automatically detect the stream and display it. 
When I tried to do the same on Ubuntu, I got first a message saying that it could not find `libgstreamer-1.0-0.dll`. I found that adding 
```xml
<dllmap dll="libgstreamer-1.0-0.dll" target="libgstreamer-1.0.so" os="!windows"/>
<dllmap dll="libgstapp-1.0-0.dll" target="libgstapp-1.0.so" os="!windows"/>
```
to `/etc/mono/config` solved that problem but then I got that error 
```csharp
INFO MissionPlanner.MainV2 - AutoConnect.NewVideoStream udpsrc port=5600 buffer-size=90000 ! application/x-rtp ! rtph264depay ! avdec_h264 ! queue leaky=2 ! videoconvert ! video/x-raw,format=BGRA ! appsink name=outsink sync=false
INFO MissionPlanner.Utilities.GStreamer - Found gstreamer /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0.1405.0
INFO MissionPlanner.Utilities.GStreamer - GStreamer 1.14.5.0
ERROR MissionPlanner.MainV2 - System.Runtime.InteropServices.MarshalDirectiveException: Marshalling of type object is not implemented
  at (wrapper managed-to-native) MissionPlanner.Utilities.GStreamer+NativeMethods.gst_app_sink_set_callbacks(intptr,MissionPlanner.Utilities.GStreamer/GstAppSinkCallbacks,int&,object)
  at MissionPlanner.Utilities.GStreamer.StartA (System.String stringpipeline) [0x0020c] in <b717f233c8d44cd1998b86fdc96361e3>:0 
  at MissionPlanner.MainV2+<>c.<OnLoad>b__114_1 (System.Object sender, System.String gststring) [0x000bf] in <23399faeeb3d49e2b220e5f0cb568dae>:0
```
It seemed to say that Mono could not use the `object` type in https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/GStreamer.cs#L182-L184 so I tried to modify it with 
```csharp
public static extern void gst_app_sink_set_callbacks(IntPtr appsink, GstAppSinkCallbacks callbacks,
                ref int testdata, GDestroyNotify p);
public delegate void GDestroyNotify(IntPtr user_data);
``` 
from information given [here](https://gstreamer.freedesktop.org/documentation/applib/gstappsink.html?gi-language=c#gst_app_sink_set_callbacks), but then Mission Planner was crashing with a segfault. 

In the end, I found that just removing 
https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/GStreamer.cs#L182-L184 and https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/GStreamer.cs#L422 solved the problem and it was still working also on Windows.

#### Version
1.3.72

#### OS
Ubuntu 18.04 with Mono 6.10.0.104

meee1, 2020-07-26:
could you test this change
    public static extern void gst_app_sink_set_callbacks(IntPtr appsink, GstAppSinkCallbacks callbacks,
                ref int user_data, ref int notify);

so i looks at 
ERROR MissionPlanner.MainV2 - System.Runtime.InteropServices.MarshalDirectiveException: Marshalling of type object is not implemented
  at (wrapper managed-to-native) MissionPlanner.Utilities.GStreamer+NativeMethods.gst_app_sink_set_callbacks(intptr,MissionPlanner.Utilities.GStreamer/GstAppSinkCallbacks,int&,object)

the object is the last element being pushed to that method, so if we better define that object, which is a GDestroyNotify, which is actually a pointer. i think it should work

lebarsfa, 2020-07-26:
This solves the exception error, but I still have a segfault after :
```csharp
INFO MissionPlanner.MainV2 - AutoConnect.NewVideoStream udpsrc port=5600 buffer-size=90000 ! application/x-rtp ! rtph264depay ! avdec_h264 ! queue leaky=2 ! videoconvert ! video/x-raw,format=BGRA ! appsink name=outsink sync=false
INFO MissionPlanner.Utilities.GStreamer - Found gstreamer /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0.1405.0
INFO MissionPlanner.Utilities.GStreamer - GStreamer 1.14.5.0
INFO MissionPlanner.Utilities.GStreamer - graphviz of pipeline is at /tmp/pipeline.dot
HUD 1 hz drawtime 0 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 0/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 1/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 2/60
HUD 1 hz drawtime 2 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 3/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 4/60
HUD 1 hz drawtime 0 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 5/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 6/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 7/60
HUD 1 hz drawtime 7 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 8/60
HUD 1 hz drawtime 1 gl True

=================================================================
	Native Crash Reporting
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================
```
That segfault seems to happen when one of the callback is called (I used some `Sleep(60000)` at different places to get an idea on what caused it). I noticed that if I replace `+=` by `=` in https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/GStreamer.cs#L414 and remove https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/GStreamer.cs#L418 everything works except that in reality `new_buffer()` callback does not seem to be called on Ubuntu 18.04 since when I add a `log.Info("XXX")` inside, it does not appear in the terminal (contrary to other `log.Info()`) while on Windows it appears as expected in `C:\ProgramData\Mission Planner\MissionPlanner.log`. I tried also with Ubuntu 20.04 (which probably has a more recent gstreamer) and I got the same behavior as Ubuntu 18.04, I tried also gstreamer 1.16.2 on Windows and I got the same behavior as with gstreamer 1.12.4 on Windows.

Therefore, it seems a Linux-specific problem, but at the moment I can mitigate it by changing the 2 lines above in addition to what you suggested (but note that I do not know much C#). Also, about the message I mentioned in the beginning of the issue
> I got first a message saying that it could not find `libgstreamer-1.0-0.dll`. I found that adding
> 
> ```
> <dllmap dll="libgstreamer-1.0-0.dll" target="libgstreamer-1.0.so" os="!windows"/>
> <dllmap dll="libgstapp-1.0-0.dll" target="libgstapp-1.0.so" os="!windows"/>
> ```
> 
> to `/etc/mono/config` solved that problem

maybe it could be good to update the message https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/GStreamer.cs#L361-L362 with something like "Please verify permissions or update DllMaps in /etc/mono/config (Linux) or /Library/Frameworks/Mono.framework/Versions/Current/etc/mono/config (Mac OS)" to give the user a hint.

lebarsfa, 2020-07-26:
Unfortunately, now for some reason I really need to remove https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/GStreamer.cs#L422 to avoid a segfault on Ubuntu...

meee1, 2020-07-27:
did you self compile gstreamer? or apt-get ?

meee1, 2020-07-27:
the reason i ask is
https://ubuntu.pkgs.org/18.04/ubuntu-updates-main-amd64/libgstreamer1.0-0_1.14.5-0ubuntu1~18.04.1_amd64.deb.html
includes /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
which i have the mono config for already. but you appear to have the non .0 file?

lebarsfa, 2020-07-27:
Yes, I have that version from apt and the files 
```bash
ls -l /usr/lib/x86_64-linux-gnu/libgstreamer-1.0*
lrwxrwxrwx 1 root root      28 juil.  3  2019 /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so -> libgstreamer-1.0.so.0.1405.0
lrwxrwxrwx 1 root root      28 mars  26 23:09 /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0 -> libgstreamer-1.0.so.0.1405.0
-rw-r--r-- 1 root root 1284520 juil.  3  2019 /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0.1405.0
```
OK, I did not know that there was https://github.com/ArduPilot/MissionPlanner/blob/6f9a25d23cd89cdfe3ba9239cb8b437ee078507f/ExtLibs/Utilities/MissionPlanner.Utilities.config#L2
I quickly checked, but for some reason it does not seem to be taken into account if I put 
```xml
<dllmap dll="libgstreamer-1.0-0.dll" target="libgstreamer-1.0.so" os="!windows"/>
<dllmap dll="libgstapp-1.0-0.dll" target="libgstapp-1.0.so" os="!windows"/>
```
inside and remove those lines from `/etc/mono/config`... Maybe there is something specific to do to take it into account, I just rebuilt?

meee1, 2020-07-27:
found the issue why its not included
Mono: Config attempting to parse: '/mnt/c/Users/michael/Source/Repos/MissionPlanner/bin/Debug/net461/MissionPlanner.Utilities.dll.config'.
Mono: Config attempting to parse: '/etc/mono/assemblies/MissionPlanner.Utilities/MissionPlanner.Utilities.config'.

file name issue. need extra .dll in there

meee1, 2020-07-27:
im pushing a new beta MP atm, that i hope will just work.

fixs the callback and the libgstreamer filename. t

lebarsfa, 2020-07-27:
Many thanks, the beta works!

Note that on Ubuntu, the output is
```csharp
INFO MissionPlanner.MainV2 - AutoConnect.NewVideoStream udpsrc port=5600 buffer-size=90000 ! application/x-rtp ! rtph264depay ! avdec_h264 ! queue leaky=2 ! videoconvert ! video/x-raw,format=BGRA ! appsink name=outsink sync=false
INFO MissionPlanner.Utilities.GStreamer - Found gstreamer /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0.1405.0
INFO MissionPlanner.Utilities.GStreamer - GStreamer 1.14.5.0

** (cli:3906): CRITICAL **: 13:14:19.413: gst_app_sink_set_callbacks: assertion 'callbacks != NULL' failed
INFO MissionPlanner.Utilities.GStreamer - graphviz of pipeline is at /tmp/pipeline.dot
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 0/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 1/60
HUD 1 hz drawtime 5 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 2/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 3/60
HUD 1 hz drawtime 2 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 4/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 5/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 6/60
HUD 1 hz drawtime 0 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 7/60
HUD 1 hz drawtime 1 gl True
INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 8/60
HUD 1 hz drawtime 1 gl True
HUD 20 hz drawtime 1 gl True
HUD 26 hz drawtime 1 gl True
HUD 25 hz drawtime 1 gl True
HUD 26 hz drawtime 1 gl True
```
while on Windows it is
```csharp
2020-07-27 13:01:23,096  INFO MissionPlanner.MainV2 - AutoConnect.NewVideoStream udpsrc port=5600 buffer-size=90000 ! application/x-rtp ! rtph264depay ! avdec_h264 ! queue leaky=2 ! videoconvert ! video/x-raw,format=BGRA ! appsink name=outsink sync=false (:0) [33]
2020-07-27 13:01:23,098 DEBUG MissionPlanner.Program - TypeResolve Failed: Mono.Runtime from MissionPlanner.Utilities, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [33]
2020-07-27 13:01:24,228  INFO MissionPlanner.Utilities.GStreamer - Found gstreamer C:\gstreamer\1.0\x86_64\bin\libgstreamer-1.0-0.dll (:0) [33]
2020-07-27 13:01:24,285  INFO MissionPlanner.Utilities.GStreamer - GStreamer 1.12.4.0 (:0) [33]
2020-07-27 13:01:24,343  INFO MissionPlanner.Utilities.GStreamer - graphviz of pipeline is at C:\Users\Administrator\AppData\Local\Temp\pipeline.dot (:0) [33]
2020-07-27 13:01:25,847  INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 0/60 (:0) [gstreamer]
2020-07-27 13:01:26,848  INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 1/60 (:0) [gstreamer]
2020-07-27 13:01:27,848  INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 2/60 (:0) [gstreamer]
2020-07-27 13:01:28,848  INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 3/60 (:0) [gstreamer]
2020-07-27 13:01:29,849  INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 4/60 (:0) [gstreamer]
2020-07-27 13:01:30,849  INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 5/60 (:0) [gstreamer]
2020-07-27 13:01:31,849  INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 6/60 (:0) [gstreamer]
2020-07-27 13:01:32,850  INFO MissionPlanner.Utilities.GStreamer - failed gst_app_sink_try_pull_sample 7/60 (:0) [gstreamer]
2020-07-27 13:01:33,097  INFO MissionPlanner.Utilities.GStreamer - new_preroll (:0) [41]
```
which seems to mean that the `new_preroll()` callback is correctly called on Windows while it is not on Ubuntu, but it does not seem to cause visible problems so I propose to close this issue.

Also, did you see my other issue #2436 about MJPEG on Ubuntu, which seems to have an easy fix (especially the first part)?

hackoma, 2021-09-19:
can you please @lebarsfa  @meee1  send me the ".exe" file, i cant build the whole mission planner. and also have the same problem with Ubuntu20.04. and MP 1.3.75
[image]

meee1, 2021-09-19:
Do you have stream installed on your system?

hackoma, 2021-09-19:
Yes @meee1 
I have installed gstreamer in ubuntu 20.04 using apt
also i have this file /usr/lib/x86_64-linux-gnu/
libgstreamer-1.0.so
libgstreamer-1.0.so.0
libgstreamer-1.0.so.0.1602.0

lebarsfa, 2021-09-19:
I confirm I have the same error message for Mission Planner 1.3.75 on Ubuntu 20.04 or 18.04. However Mission Planner 1.3.74 (http://firmware.ardupilot.org/Tools/MissionPlanner/MissionPlanner-1.3.74.zip) appears to work fine.

lebarsfa, 2021-09-19:
It seems due to https://github.com/ArduPilot/MissionPlanner/commit/71aa0f8fcb5b5b7b89a07c29c95610e49ecd749d#diff-a791f2d9d060eb3e23e7e0765f713363c513ae7428b453a3e543b0980e3bdcff : MissionPlanner.Utilities.dll.config vs MissionPlanner.Utilities.config in ExtLibs/Utilities/MissionPlanner.Utilities.csproj .

hackoma, 2021-09-19:
unfortunately i'm using 1.3.74 but also No video !
when i'm starting this line 
`gst-launch-1.0 videotestsrc pattern=ball ! x264enc ! h264parse ! rtph264pay ! udpsink host=127.0.0.1 port=5600` 
i have this in MP terminal
[image]

lebarsfa, 2021-09-19:
I have the same types of messages, however in the end the video appears (`failed gst_app_sink_try_pull_sample` messages count up to around 10). Maybe it could be another app using the UDP port 5600, e.g. the `failed gst_app_sink_try_pull_sample` messages would stay up to 60 and say `Gstreamer Exit` if I launch a 2nd instance of Mission Planner, which would never show the video while the first instance works well...

hackoma, 2021-09-20:
i'm making sure there are no another application using this port " by display all listening udp port using ss -lnu " 
So can you please @lebarsfa  share your /etc/mono/config file

lebarsfa, 2021-09-20:
Using live CD `ubuntu-20.04.3-desktop-amd64.iso` (ensure` Community software (universe)` is checked in `Software & Updates` GUI) : 

```bash
sudo apt update
sudo apt install mono-complete gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly x264
gst-launch-1.0 videotestsrc pattern=ball ! x264enc ! h264parse ! rtph264pay ! udpsink host=127.0.0.1 port=5600
```

```bash
wget https://firmware.ardupilot.org/Tools/MissionPlanner/MissionPlanner-1.3.74.zip
# or temporarily corrected 1.3.75 version https://www.ensta-bretagne.fr/lebars/Share/MissionPlanner-linux.zip
unzip -q -o MissionPlanner-1.3.74.zip -d MissionPlanner
cd MissionPlanner
mono MissionPlanner.exe
```

FYI `/etc/mono/config`, but I think it is left to default : 
```xml
<configuration>
	<dllmap dll="i:cygwin1.dll" target="libc.so.6" os="!windows" />
	<dllmap dll="libc" target="libc.so.6" os="!windows"/>
	<dllmap dll="intl" target="libc.so.6" os="!windows"/>
	<dllmap dll="intl" name="bind_textdomain_codeset" target="libc.so.6" os="solaris"/>
	<dllmap dll="libintl" name="bind_textdomain_codeset" target="libc.so.6" os="solaris"/>
	<dllmap dll="libintl" target="libc.so.6" os="!windows"/>
	<dllmap dll="i:libxslt.dll" target="libxslt.so" os="!windows"/>
	<dllmap dll="i:odbc32.dll" target="libodbc.so.2" os="!windows"/>
	<dllmap dll="i:odbc32.dll" target="libiodbc.dylib" os="osx"/>
	<dllmap dll="oci" target="libclntsh.so" os="!windows"/>
	<dllmap dll="db2cli" target="libdb2_36.so" os="!windows"/>
	<dllmap dll="MonoPosixHelper" target="$mono_libdir/libMonoPosixHelper.so" os="!windows" />
	<dllmap dll="System.Native" target="$mono_libdir/libmono-native.so" os="!windows" />
	<dllmap dll="System.Net.Security.Native" target="$mono_libdir/libmono-native.so" os="!windows" />
	<dllmap dll="System.Security.Cryptography.Native.Apple" target="$mono_libdir/libmono-native.so" os="osx" />
	<dllmap dll="libmono-btls-shared" target="$mono_libdir/libmono-btls-shared.so" os="!windows" />
	<dllmap dll="i:msvcrt" target="libc.so.6" os="!windows"/>
	<dllmap dll="i:msvcrt.dll" target="libc.so.6" os="!windows"/>
	<dllmap dll="sqlite" target="libsqlite.so.0" os="!windows"/>
	<dllmap dll="sqlite3" target="libsqlite3.so.0" os="!windows"/>
	<dllmap dll="libX11" target="libX11.so.6" os="!windows" />
	<dllmap dll="libgdk-x11-2.0" target="libgdk-x11-2.0.so.0" os="!windows"/>
	<dllmap dll="libgdk_pixbuf-2.0" target="libgdk_pixbuf-2.0.so.0" os="!windows"/>
	<dllmap dll="libgtk-x11-2.0" target="libgtk-x11-2.0.so.0" os="!windows"/>
	<dllmap dll="libglib-2.0" target="libglib-2.0.so.0" os="!windows"/>
	<dllmap dll="libgobject-2.0" target="libgobject-2.0.so.0" os="!windows"/>
	<dllmap dll="libgnomeui-2" target="libgnomeui-2.so.0" os="!windows"/>
	<dllmap dll="librsvg-2" target="librsvg-2.so.2" os="!windows"/>
	<dllmap dll="libXinerama" target="libXinerama.so.1" os="!windows" />
	<dllmap dll="libasound" target="libasound.so.2" os="!windows" />
	<dllmap dll="libcairo-2.dll" target="libcairo.so.2" os="!windows"/>
	<dllmap dll="libcairo-2.dll" target="libcairo.2.dylib" os="osx"/>
	<dllmap dll="libcups" target="libcups.so.2" os="!windows"/>
	<dllmap dll="libcups" target="libcups.dylib" os="osx"/>
	<dllmap dll="i:kernel32.dll">
		<dllentry dll="__Internal" name="CopyMemory" target="mono_win32_compat_CopyMemory"/>
		<dllentry dll="__Internal" name="FillMemory" target="mono_win32_compat_FillMemory"/>
		<dllentry dll="__Internal" name="MoveMemory" target="mono_win32_compat_MoveMemory"/>
		<dllentry dll="__Internal" name="ZeroMemory" target="mono_win32_compat_ZeroMemory"/>
	</dllmap>
	<dllmap dll="gdiplus" target="libgdiplus.so.0" os="!windows"/>
	<dllmap dll="gdiplus.dll" target="libgdiplus.so.0"  os="!windows"/>
	<dllmap dll="gdi32" target="libgdiplus.so.0" os="!windows"/>
	<dllmap dll="gdi32.dll" target="libgdiplus.so.0" os="!windows"/>
</configuration>
```

lebarsfa, 2021-09-27:
@meee1 In case you missed it : 

> It seems due to [71aa0f8#diff-a791f2d9d060eb3e23e7e0765f713363c513ae7428b453a3e543b0980e3bdcff](https://github.com/ArduPilot/MissionPlanner/commit/71aa0f8fcb5b5b7b89a07c29c95610e49ecd749d#diff-a791f2d9d060eb3e23e7e0765f713363c513ae7428b453a3e543b0980e3bdcff) : MissionPlanner.Utilities.dll.config vs MissionPlanner.Utilities.config in ExtLibs/Utilities/MissionPlanner.Utilities.csproj .

lebarsfa, 2021-09-28:
OK many thanks!

closed 2021-09-28 by lebarsfa