ArduPilot/MissionPlanner #2442: Purpose of the getParamPoll() function

Vassiliev99, 2020-07-31:
Hi! Why do we need getParamPoll() [here](https://github.com/ArduPilot/MissionPlanner/blob/master/MainV2.cs#L2790)?
```
if (!port.giveComport)
{
	try
	{
		// poll only when not armed
		if (!port.MAV.cs.armed)
		{
			port.getParamPoll();
			port.getParamPoll();
		}
	}
	catch
	{
	}
}
```
Sometimes it causes problems with loading parameters on connection. I've removed this piece of code and everything worked fine and without getting "Already got param" message.

meee1, 2020-07-31:
it polls for params on every mavlink endpoint.
also "Already got param" isnt realy an issue in this context

Vassiliev99, 2020-07-31:
> it polls for params on every mavlink endpoint.
> also "Already got param" isnt realy an issue in this context

But what is the reason to poll for params on every mavlink endpoint? What will happen if we would not do it?