ArduPilot/MissionPlanner #2444: Radar view for OBSTACLE_DISTANCE message

rmackay9, 2020-08-10:
With the ongoing work to improve obstacle avoidance using the Intel 435i, AP is receiving OBSTACLE_DISTANCE messages.  It would be great if we could display the output of these messages on the radar view.  This would make both development and use of these sensors easier because the user could more easily see what output is coming from the sensor.  For example it's helpful in determining if the sensor is aligned forward properly.

thien94, 2020-08-10:
Here is an example script that I created to test the OBSTACLE_DISTANCE (OD) message: https://github.com/thien94/vision_to_mavros/blob/t265_and_d435/scripts/AP_test_obstacle_distances.py

1. Change the connection parameters to the FCU: [string and baudarate](https://github.com/thien94/vision_to_mavros/blob/t265_and_d435/scripts/AP_test_obstacle_distances.py#L30-L32)
2. The script will then send the OD message at a [pre-defined frequency](https://github.com/thien94/vision_to_mavros/blob/t265_and_d435/scripts/AP_test_obstacle_distances.py#L37).
3. Modify the configurations for the OD message [here](https://github.com/thien94/vision_to_mavros/blob/t265_and_d435/scripts/AP_test_obstacle_distances.py#L78-L108).

Below is the result by changing just the simulated FOV ([this param](https://github.com/thien94/vision_to_mavros/blob/t265_and_d435/scripts/AP_test_obstacle_distances.py#L96)):
- angle_start = -39.5, angle_end = 39.5 (or FOV = 78 degrees, similar to real D435 camera): only 2 arcs appeared on MP's Proximity View
[image]
- angle_start = -39.5, angle_end < 69 will also result in 2 arcs.
- angle_start = -39.5, angle_end > 70 will result to 3 arcs:
[image]

meee1, 2020-08-10:
ive added code to the new beta already, have not tested it yet though

meee1, 2020-08-10:
[image]

tested. and slight change to background