ArduPilot/MissionPlanner #2450: Cannot complete the device driver installation wizard

cnpcshangbo, 2020-08-25:
When I install Mission Planner, I got a message said: "Cannot complete the device driver installation wizard".
[image]
After the installation, I tried to connect a serial port, however, it is recognized as Microsoft serial ballpoint.
[image]
I have tried to change register as mentioned in https://stackoverflow.com/questions/9226082/device-misdetected-as-serial-mouse
However, I am still refused to access the serial port. I got a message: Access to the port 'COM3' is denied.
[image]
Hope to get some suggestions. Thanks.

meee1, 2020-08-25:
I would reboot the VM and try again. Most likely the miss detection is causing it

cnpcshangbo, 2020-08-27:
@meee1 Thank you. Rebooting the computer solved this issue.

closed 2020-08-27 by cnpcshangbo