ArduPilot/MissionPlanner #2455: Problem to change flight modes

Giacomo19999, 2020-09-01:
#### Issue details
_Please describe the problem, or desired feature_

Currently, I see changing channel 5 value in Mission Planner, in particular in the Radio Calibration menu, but the flight modes in Flight Modes menu don't change and the Current PWM is always 0. I don't know if there is a relation whit this problem and the error PreArm that I have permanently. 

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
hex
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
APM 2.8
#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2020-09-01:
If pwm is zero, then it's not receiving that channel. Are your other channels working? 

Please consider yourself your hardware. The 8bit APM 2.8 board was deprecated many years ago and runs very old firmware.

Giacomo19999, 2020-09-01:
Thank you for the reply. In Radio Calibration menu I see changing channel 5
pwm when I control the switch, but in Flight Modes menu never. Could you
recommend one at a low price?