ArduPilot/MissionPlanner #2456: mono submodule has broken (AssemblyInfo.cs does not exitst for the System.Runtime.Serialization.Formatters.Soap)

darkcircle, 2020-09-03:
#### Issue details
ExtLibs/mono/mcs/class/System.Runtime.Serialization.Formatters.Soap/Assembly does not exist.

#### Version
_Current Upstream_

#### Platform
[v] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_Not applicable_

#### Hardware type
_Not applicable_

#### Logs
```
CSC     [net_4_x-win32] System.Data.dll
D:\dev\mp-master\ExtLibs\mono\external\corefx\src\System.Data.SqlClient\src\System\Data\SqlClient\SqlConnection.cs(46,23): warning CS0108: 'SqlConnection._suppressStateChangeForReconnection' hides inhe
rited member 'DbConnection._suppressStateChangeForReconnection'. Use the new keyword if hiding was intended.
D:\dev\mp-master\ExtLibs\mono\external\corefx\src\System.Data.SqlClient\src\System\Data\SqlClient\SqlCommand.cs(3909,23): warning CS0108: 'SqlCommand.CancelIgnoreFailure()' hides inherited member 'DbCo
mmand.CancelIgnoreFailure()'. Use the new keyword if hiding was intended.
Creating ../../build/deps/win32_net_4_x_System.ComponentModel.DataAnnotations_test.dll.makefrag ...
CSC     [net_4_x-win32] System.ComponentModel.DataAnnotations.dll
// Error(s) in file D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap\System.Runtime.Serialization.Formatters.Soap.dll.sources:
Directory does not exist: 'D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap\Assembly'
// gensources produced 1 error(s) and a set of 7 filename(s)
// Invoked with 'D:\dev\mp-master\ExtLibs\mono\mcs\class\lib\build\gensources.exe --strict --platformsdir:./../../build ../../build/deps/win32_net_4_x__System.Runtime.Serialization.Formatters.Soap.dll.
sources System.Runtime.Serialization.Formatters.Soap.dll win32 net_4_x'
// Working directory was 'D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap'
Creating ../../build/deps/win32_net_4_x_System.Runtime.Serialization.Formatters.Soap_test.dll.makefrag ...
// Error(s) in file D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap\System.Runtime.Serialization.Formatters.Soap.dll.sources:
Directory does not exist: 'D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap\Assembly'
// gensources produced 1 error(s) and a set of 7 filename(s)
// Invoked with 'D:\dev\mp-master\ExtLibs\mono\mcs\class\lib\build\gensources.exe --strict --platformsdir:./../../build ../../build/deps/win32_net_4_x__System.Runtime.Serialization.Formatters.Soap.dll.
sources System.Runtime.Serialization.Formatters.Soap.dll win32 net_4_x'
// Working directory was 'D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap'
// Error(s) in file D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap\System.Runtime.Serialization.Formatters.Soap.dll.sources:
Directory does not exist: 'D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap\Assembly'
// gensources produced 1 error(s) and a set of 7 filename(s)
// Invoked with 'D:\dev\mp-master\ExtLibs\mono\mcs\class\lib\build\gensources.exe --strict --platformsdir:./../../build ../../build/deps/win32_net_4_x__System.Runtime.Serialization.Formatters.Soap.dll.
sources System.Runtime.Serialization.Formatters.Soap.dll win32 net_4_x'
// Working directory was 'D:\dev\mp-master\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap'
make[9]: *** '../../build/deps/win32_net_4_x__System.Runtime.Serialization.Formatters.Soap.dll.response'에서 필요한 '../../build/deps/win32_net_4_x__System.Runtime.Serialization.Formatters.Soap.dll.sources' 타겟을 만들 규칙이 없습니다.  멈춤.
make[8]: *** [../../build/rules.make:220: do-all] 오류 2
make[7]: *** [../build/rules.make:258: System.Runtime.Serialization.Formatters.Soap] 오류 2
make[6]: *** [../build/rules.make:241: all-recursive] 오류 1
make[5]: *** [build/rules.make:241: all-recursive] 오류 1
make[4]: *** [Makefile:58: profile-do--net_4_x--all] 오류 2
make[3]: *** [Makefile:54: profiles-do--all] 오류 2
make[2]: *** [Makefile:712: all-mcs] 오류 2
make[2]: 디렉터리 '/cygdrive/d/dev/mp-master/ExtLibs/mono/runtime' 나감
make[1]: *** [Makefile:605: all-recursive] 오류 1
make[1]: 디렉터리 '/cygdrive/d/dev/mp-master/ExtLibs/mono' 나감
make: *** [Makefile:533: all] 오류 2
```

darkcircle, 2020-09-03:
Currently, project is broken if I clones freshly. there are some item should be in for referring symbol.
I was doing reproduce for making successful build, I failed to build Mission Planner until now (night) from the morning.

Could you check again from the fresh source tree came from github repo?

meee1, 2020-09-13:
this is now fixed. the extra projects are included by default. they where always there, just not in the MP solution

closed 2020-09-13 by meee1