ArduPilot/MissionPlanner #2457: Mission Planner changes flight mode without user input

AntonSlooten, 2020-09-04:
#### Issue details
Mission Planner alters flight mode arbitrarily without any user input or knowledge Aside from the obvious "Flight mode changed to 'X'"
It is also doing this using an obsoleted message #11 - SET_MODE, rather than the correct #76 - COMMAND_LONG  message.

I have attached logs of both the offending #11 messages to change mode and the #76 messages from the log files. You can see the mode is changed at reasonably high frequencies - not one of these is requested by the user.

#### Version
Mission planner: 1.3.7530.26934
Arducopter 4.0.3

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quad X8

#### Hardware type
Cube Orange

#### Logs

Here is an image of mission planner changing mode commanded:
 [image]

Here are logs off our onboard computer of the modes being requested to change through mavlink message #11:
[SET_MODE.log](https://github.com/ArduPilot/MissionPlanner/files/5172320/SET_MODE.log)

Here is all instances of #76
[COMMAND_LONG.log](https://github.com/ArduPilot/MissionPlanner/files/5172324/COMMAND_LONG.log)

meee1, 2020-09-04:
i need either tlogs or ardupilot logs. as MP wouldnt be changing the modes without some form of user initiation

meee1, 2020-09-04:
regarding the depreciated
https://github.com/mavlink/qgroundcontrol/issues/8776
also MP current beta actually sends both, as it needs to be backwards compatible

rmackay9, 2020-10-19:
I have also seen this issue and it seemed to be caused by the flight mode drop-down feature in beta MP.  My understanding is that @meee1 has disabled this drop-down in the latest MP and in stable.

[image]

This is a terrible bug actually.

meee1, 2020-10-19:
the cause is a race condition between the selection in the dropdown and the mode reported from the autopilot. current stable release has this disabled by default