ArduPilot/MissionPlanner #2458: Desired: Terrain following toggle

JonReacher, 2020-09-04:
It would be helpful to have a dropdown on/off selector for the TERRAIN_FOLLOW parameter on the Data/Actions screen. Although in general it may be good to keep it enabled for safety at low to moderate altitudes, RTL, etc, it is better for endurance to simply fly level for most of the flight. 

For commercial purposes where we have blocked off the config pages it would be nice to have this exposed to set on the fly, as I don't see any real downside concerns, as long as the tiles are loaded anyway (and maybe disabled if TERRAIN_ENABLE is 0?). It could be an example plugin that could be disabled for those who don't want it, which may be ideal.

meee1, 2020-09-04:
is it possible to add plugin to MP that would add this type of function. give you have block off config already, are you using a custom version?

JonReacher, 2020-09-04:
No, stock version. Latest stable. I'm guessing you mean it is possible? I would be all over it myself if I had any idea of how you guys do what you do! This would be a big add for those looking for that balance of safety/ endurance for longer flights and distances.