ArduPilot/MissionPlanner #2460: gstreamer did not work

cyhly, 2020-09-07:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2020-09-07:
please provide more information.

cyhly, 2020-09-07:
Sorry，i want to know how to use gstreamer with missionplanner?

meee1, 2020-09-07:
best to ask this question on the ardupilot forum in the mission planner section.

closed 2020-09-07 by meee1

cyhly, 2020-09-09:
i am sorry，i ask this question on the ardupilot forum in the mission planner section.but no reply.
I downloaded the GStreamer component on C disk.
but when i use gstreamer with Right-click hud.
there was no video shown.
I really need this right now.
what should i do?

meee1, 2020-09-09:
no idea where your post is
should be here
https://discuss.ardupilot.org/c/ground-control-software/mission-planner/