ArduPilot/MissionPlanner #2464: MAVFtp: crc error

IamPete1, 2020-09-12:
When downloading files via FTP, both to a real vehicle and SILT, I get a CRC error (MP 1.3.73 build 1.3.558.11943). Also reported  in https://github.com/ArduPilot/MissionPlanner/issues/2422#issuecomment-652200697.

[image]

MAVProxy will download the same file with `ftp get` without error. The file is actually received fine, it just ends up with some random characters appended. Sending files from MP to AP works as expected. The same error is seen with both `Download` and `Download burst`.

Interestingly param download via MAVFtp works fine. I came across this while trying to track down a issue with the scripting REPL,  it works with SILT but not on real hardware, suspect that is a AP bug rather than a MP one tho.

meee1, 2020-09-13:
do you have the tlog for this?

IamPete1, 2020-09-13:
I have put it on [my drive](https://drive.google.com/drive/folders/1jpTm6blDFDA27H3uvAGsNsMgGvBUeLm5?usp=sharing
). 

There is tlog, rlog and the file as it was receved with the extra junk. The orignal file is [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/simple_loop.lua).

Thanks

meee1, 2020-09-13:
is the lua from a download?
i see this
[image]

IamPete1, 2020-09-13:
Yeah, that is the stuff that gets appended.

meee1, 2020-09-13:
ok, ill start looking there.

IamPete1, 2020-09-13:
Notepad++ does a better job at displaying them than VS Code, so hadn't considered it before, but I checked, you get more `NULL`s for bigger files, so its not just a constant number at the end. Doesn't seem directly related to the file size tho as far as I can tell.

I also forgot to mention that this used to work fine, I don't recall when exactly tho.

meee1, 2020-09-13:
i think i now why, its to do with the reported file size, and virtual files.... i will need to look harder.

i can replicate it in sitl, so makes it easier to find

meee1, 2020-09-13:
new beta in 10 mins

IamPete1, 2020-09-13:
Can confirm is now fixed, thank you!

closed 2020-09-13 by IamPete1