ArduPilot/MissionPlanner #2466: Scripting REPL: works in SITL but not with real hardware

IamPete1, 2020-09-14:
I will preface this with saying that I'm not sure this is a MP issue at all, could be a AP file system issue.

The Scripting REPL works perfectly with AP SILT in all version of MP since it was added. However I cannot get it to work with real hardware (a Cube Orange). I have tested with MP current beta, 1.3.71 and 1.3.72. I have also tested original REPL [PR branch](https://github.com/ArduPilot/ardupilot/pull/13547) of AP with the same results.

That makes it seem like a AP issue except that @WickedShell and I are sure we have had it working with real hardware at some point in the past, probably several months ago when the feature was first introduced. It is likely that we were both on random beta MP builds at the time.

Unfortunately we cant find the build of MP where it worked to allow us to do a direct comparison of the original AP PR branch and current AP master (of course this assumes that the original AP PR branch does in-fact work).

Basically I'm a bit lost tracking this down from the AP side.

WickedShell, 2020-09-14:
To clarify this a bit further, if I test with a near master build (some GPS driver changes only) on a Cube Orange the MP version I have here hangs and doesn't manage to open the repl session. But if I restart MP I can directly FTP the in/out files for the REPL and they change appropriately, which leads me to guess it's something on the MP side, but I'm not sure where.

IamPete1, 2020-11-20:
This is now fixed in beta MP

closed 2020-11-20 by IamPete1

meee1, 2020-11-20:
I believe this was a /repl/. Vs /APM/repl/