ArduPilot/MissionPlanner #2469: MP crash on ubuntu 18.04 with latest  mono

anbello, 2020-09-17:
#### Issue details
Mission Planner crash on ubuntu after laste mono update.
Before latest mono update it worked good,

#### Version
Ubuntu 18.04

MissionPlanner-1.3.73

$ mono --version

```
Mono JIT compiler version 6.12.0.90 (tarball Fri Sep  4 14:01:23 UTC 2020)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          yes(610)
	Suspend:       hybrid
	GC:            sgen (concurrent by default)
```

$ mono MissionPlanner.exe

```
If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35 

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
To fix any filename case issues under mono use    export MONO_IOMAP=drive:case
Data Dir /home/andrea/Mission Planner/
Log Dir /home/andrea/Mission Planner/logs
Running Dir /home/andrea/Programmi/MissionPlanner-1.3.73/
User Data Dir /home/andrea/Mission Planner/
at MissionPlanner.exe.Program.Main  (IL offset: 0x1)
at MissionPlanner.exe.Program.Start  (IL offset: 0x107)
at System.Windows.Forms.dll.Application.EnableVisualStyles  (IL offset: 0x6)
Message: The type initializer for 'System.Windows.Forms.XplatUI' threw an exception.
CustomMessageBox.Show
exception inside UnhandledException handler: (null) assembly:/usr/lib/mono/4.5/mscorlib.dll type:Exception member:(null)

[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: The type initializer for 'System.Windows.Forms.XplatUI' threw an exception. ---> System.MissingMethodException: Method not found: int Mono.Unix.Native.Syscall.fcntl(int,Mono.Unix.Native.FcntlCommand,int)
  at System.Windows.Forms.XplatUIX11..ctor () [0x00077] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.XplatUIX11.GetInstance () [0x00019] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.XplatUI..cctor () [0x000c0] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
   --- End of inner exception stack trace ---
  at System.Windows.Forms.Application.EnableVisualStyles () [0x00006] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at MissionPlanner.Program.Start (System.String[] args) [0x00107] in <231ad899de864fa081cf231efbd9281c>:0 
  at MissionPlanner.Program.Main (System.String[] args) [0x00001] in <231ad899de864fa081cf231efbd9281c>:0 

```

meee1, 2020-09-17:
this is a mono bug. i would suggest either an up or downgrade on mono

anbello, 2020-09-17:
Thanks I can try to open an issue on mono.

meee1, 2020-09-17:
yeah, by the looks of it the shipped mono library is linked against a bad dll

anbello, 2020-09-17:
I just downloaded MissionPlanner-1.3.74 and I don't have the problem anymore.

closed 2020-09-19 by anbello