ArduPilot/MissionPlanner #2473: [Simple Inqury]  Adding an additional column in waypoint file?

AbhishekJoshi, 2020-09-17:
#### Issue details

I am wondering:
1) How can I add an additional column in waypoint data? Essentially I want to add a mode within the waypoint to manage a specific state so that my Copter or Rover can change cameras (or something) based on the current waypoint. This is the simplest reason why I want to add another column, such that my Copter will execute a function after reaching a waypoint.
2) Where in the mission planner software I can add such changes, including to add this column of data in the exported WP file?

#### Version
Latest

#### Platform
[X] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[ X  ] Rover
[  ] Sub

#### Airframe type
Copter Quad.

#### Hardware type
Pixhawk or PXFmini -- I can deal with the solution on the hardware level.

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2020-09-17:
i think you might be underestimating the complexity of this problem
waypoint are executed on the vehicle, so the code to do the smarts is in ardupilot. MP can send all the param fields as part of a mission, but that does not mean its saved for use on the autopilot, as it needs to conform the the locationwp struct in the autopilot.
i think you might need to start looking on how it would be done from the autopilot side first.

AbhishekJoshi, 2020-09-18:
Hi @meee1  thanks for your response. I'm aware of the complexity at the copter level. I've been doing custom modifications for Copter and Rover for quite a bit of time, but yeah this is the probably the most complicated thing I'm experimenting with.

At the moment, I actually have the column set up in the grid in Visual Studio MP, but I'm trying to figure out how to implement the dropdown **and** that it exports correctly in the WP file. Essentially I want to add a drop down of numeric value 1-5 called state, and export that as part of the WP.

I believe like you said, it is just parsing the struct correctly. I have a function that I'd like to test that manages and is ready to test, but you need mission planner to be able to export. At the vehicle level, `if state == 1: do functionOne()` is not the unknown for me.

meee1, 2020-09-18:
why are you adding another column? you cant use one of the existing ones? that would be alot easier