ArduPilot/MissionPlanner #2483: feature request: Load offline maps

MatzeR82, 2020-10-06:
#### Issue details
Is there a feature implemented for loading offline maps? If not, I recomment to add this feature. It makes the mission planning easier if you are at the test-site without a stable wifi connection.

best regards

rmackay9, 2020-10-06:
This feature does certainly exist although I'm struggling a bit to find it

jagrandle, 2020-10-06:
MatzeR82, there is a feature for saving offline maps, which is called Prefetch in this application. You can do this before you go to a test site. On the Flight Plan screen you can right click and then select Map Tool. One of the options is Prefetch. There is some more help in the Mission Planner documentation here. The videos are a bit old but you will get the ideas.

These days you can set the maximum zoom level to go down to I think. Don't go too low or it will take a long time !!

https://ardupilot.org/planner/docs/common-planning-a-mission-with-waypoints-and-events.html?highlight=prefetch#how-to-prefetch-a-stored-mission-map

meee1, 2020-10-06:
what is the origin of the offline map? i can do a guide, but not sure on what source's people are trying to use

meee1, 2020-10-06:
here is a very old link
http://vps.oborne.me/gcs/CustomImages.htm

rmackay9, 2020-10-07:
@meee1, I think @MatzeR82 is just looking to download maps so they are visible without an internet connection.

meee1, 2020-10-07:
then what @jagrandle  said is correct, use prefetch to grab the tiles prior to leaving the not internet area.

i will had google has limits on how much you can rip in 1 day. depending on how crazy you plan on going