ArduPilot/MissionPlanner #2484: [parameter docs] Description of ARSPD_FBW_MAX  wrong in Mission Planner 1.3.74 build 1.3.7563.27684

teodoryantcheff, 2020-10-06:
#### Issue details
The description of the `ARSPD_FBW_MAX ` parameter is wrong stating 
> ... and at least 50% above ARSPD_FBW_MAX  to allow ...

#### Version
MP 1.3.74 build 1.3.7563.27684

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Sub

[Link](https://github.com/ArduPilot/ardupilot_wiki/issues/2393) to where I reported this to the wiki team.

[image]

rmackay9, 2020-10-07:
The parameter descriptions come from the flight code and it looks like it's already been addressed back in January.  I'm a little surprised this hasn't made it out to an official release.
https://github.com/ArduPilot/ardupilot/pull/13153

Anyway, it's not a mission planner issue so I'll close this if that's OK.  I'll see comments reported below though even if this issue is closed.

closed 2020-10-07 by rmackay9

teodoryantcheff, 2020-10-07:
> ..this hasn't made it out to an official release...

As per me this is IS an MP issue. Maybe a build process problem, maybe something else but an MP issue nonetheless.
I've [reported that](https://github.com/ArduPilot/ardupilot_wiki/issues/2393) 9 months ago, it got fixed in the flight code almost immediately, and made it through to the wiki. There have been multiple official MP releases since and it still has the old parameter description and that leads to me think there's an issue.

rmackay9, 2020-10-07:
Maybe the MP's parameter descriptions are not up to date.  Try this:

Ctrl-F, then push the "Param gen" button (7th button in the left column) then wait a few minutes and restart MP.

This is what I see:
[image]

teodoryantcheff, 2020-10-07:
Will do. Thanks, @rmackay9 !

teodoryantcheff, 2020-10-08:
Confirmed, what @rmackay9 suggested did update the parameter descriptions. The usability and UX of that feature though are appalling.
Thanks!

rmackay9, 2020-10-08:
Ah, the parameter regeneration is also available on the Setup >> Advanced screen so that's an easier way to find it.  It might be better if it was placed on the right in the Full Parameter Tree and Full Parameter List screens.