ArduPilot/MissionPlanner #2490: Unable to edit forms in Forms Designer

PRIMETSS, 2020-10-13:
#### Issue details
When trying to visually edit the FlightData.cs form in designer, I get the blue designer error screen
[image]

Using Visual Studio 16.7.5 + .Net 4.6.1

I was able to edit some forms back in version 1.3.70 about ~4 months ago. But did notice back then some forms failed to open without errors.
Been stuck on this for days, so decided to try a few previous builds (fresh forks from repo) all no luck.
Is it an issue with a to up-to-date Visual Studio?

#### Version
1.3.74
New fork of Latest version Master Branch https://github.com/ArduPilot/MissionPlanner/commit/a9d432a204b16437391e2af5a91f959d5d9e2f7e

#### Platform
[  ] All

Thankyou, Appreciated

PRIMETSS, 2020-10-14:
Edited AGauge.cs and commented out #32 //ToolboxBitmapAttribute(typeof(AGauge)),
This skips above error but then trying to now re open FlightData Form
[image]

Read the build wiki and sure Ive done all steps... Im using VS2019 Pro latest version.
Can anyone confirm they are able to edit forms using latest VS and latest MP?

Thanks!

EosBandi, 2020-10-14:
Based on my experience the latest VS version which works flawlessly for MP is 16.7.2 There were some disruptive changes in 16.7.3 which causing issues in Forms Designer. I suggest to downgrade to 16.7.2

PRIMETSS, 2020-10-14:
Thanks for reply!
Was suspecting it might be VS's form designer that might be cause. Possibly because of the 'recent' dot core WinForms integration.
But I'm sure I had same issues with version 16.7.2 and maybe even before that, as Ive not been able to edit MP forms for at least 6-8 months (but you might be right with 16.7.2 as maybe that's about the time frame (will check release dates)). Yesterdays upgrade to 16.7.6 didn't help. I've also tried running the /safemode to disable any extensions.
Very much like to resolve it, as rather have latest VS due to dot core dev.

PRIMETSS, 2020-10-15:
Today went to laptop that was using VS 16.7.3
Copied over repo and was able to open FlightData in designer initially.
After rebuild started getting exception errors in designer again.
Drilled down into MissionPlanner.Utilities and notice (by fluke) a yellow triangle bang on System.Runtme.Loader (4.3.0) dependency
Remove and Used Nuget to add it back in as dependency and yellow bang removed.

Can now open FlightData From in designer to do edits I needed.

Did full Batch Rebuild and still seems OK at this stage.

meee1, 2020-10-15:
im using atm
[image]

PRIMETSS, 2020-10-15:
Thanks Michael!
I will re-check all ok tomorrow. But seems it might have been a missing dependency. If I can replicate and fix on Laptop with same issues tomorrow I'll be happy.
Still interested to hear from others still on VS 16.7.X to see if they are having issues as seems quite a few issues open will similar issues in the designer. Think the VS Team have been modifying the Designer to bring in dot net core.
Love to hear from anyone else on VS 16.7.X and what there experience is. As have a hunch this is going to bite me again after a rebuild....

meee1, 2020-10-15:
yeah i think the 16.7.2+ had issues. ive been running preview builds for over a year now.