ArduPilot/MissionPlanner #2497: Simulation: can't successfully start skid-rover frame

rmackay9, 2020-10-29:
For some reason I'm unable to start MP's simulator using the skid-rover frame or model type.

I would have expected that I could enter "-f skid-rover" into the "Extra command line" field but it complains, "Failed to connect to SITL instance".

Ideally we should add this very common frame to the "Model" drop-down to make it easy for users.

I strongly suspect that I've done something wrong but any help is appreciated.

meee1, 2020-11-17:
i get a

C:\Users\michael\Documents\Mission Planner\sitl>APMrover2.exe -M skid-rover
Vehicle model (skid-rover) not found

when it try sitl.  is it the correct name?

rmackay9, 2020-11-18:
I'm sorry, it's "rover-skid" :-(.  sorry about that.

closed 2020-12-01 by meee1