ArduPilot/MissionPlanner #2498: Mission Planner Issuing Unrequested Parameter Change

AuClark, 2020-10-30:
#### Issue details
On multiple occasions, we have witnessed a reset of our Hexcopters FRAME_TYPE from our custom frame designated 19 back to 0. 
The issue seems to occur on boot and only when Mission Planner Desktop is connected via UDP to Herelink GCS (Running Q Ground Control).
Investigating this issue with T Logs (With Andrew Tridgells help) we have identified that the parameter change came from Mission Planner and was definitely not prompted by the operator.

[image]

#### Version
1.3.72

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Hexacopter

#### Hardware type
Pixhawk Cube Orange

#### Logs
Link to parameters before reset [here](https://drive.google.com/file/d/1eH9nuS_tUZS3migD-W1jlAuynFQezKdh/view?usp=sharing).
Link to tLog showing reset [here](https://drive.google.com/file/d/1xgTL1fQ-_rFgqAj8n4Qog784yaIsEYH5/view?usp=sharing).
Link to parameters after reset [here](https://drive.google.com/file/d/19NFMt09ci2OlrARF-k8XrKEou29jTd5B/view?usp=sharing).

closed 2020-12-01 by meee1

AuClark, 2020-12-01:
@meee1 Hi Michael, could you please explain to me what was happening with this reset?

meee1, 2020-12-01:
opening the frame type screen when MP doesnt know about the current selected frame.

ie it gets reset to a frame known to MP.

AuClark, 2020-12-01:
@meee1 will the patch clear this up for custom frame types?

meee1, 2020-12-01:
yes it will apear as Other

AuClark, 2020-12-01:
Thanks Michael. Do I just need to update Mission planner now? Or do I need to wait for the next release?

meee1, 2020-12-01:
if you do a beta update now it should be fine