ArduPilot/MissionPlanner #2501: Parameter metadata displayed does not match the firmware on vehicle

Naterater, 2020-11-03:
#### Issue details
While flying, I intended to set the THR_FS to 2. The full parameter list indicated that 2 was a valid value to set, and the response from mission planner is that the parameter setting was successful. The behavior on the plane, however, did NOT change because it was running 4.0.6 beta 1 that could not set this parameter. The wiki and many other sources can pull the correct parameter descriptions, I would also expect Mission Planner to do the same.

#### Version
1.3.74

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Plane

#### Hardware type
Cube

rmackay9, 2020-11-03:
@meee1 I think really the parameter meta data used should depend upon the version of the vehicle code, not the version of the mission planner.

Autonomost, 2021-01-06:
@rmackay9 @meee1 I am currently digging through mission planner for find out where the Metadata file gets created. If one of you two find it before I do could you link me please? 

@Naterater - you can always delete the ParameterMetaData.xml file in your Documents/Mission Planner directory, then hit ctrl+f to open the adv menu, then hit regenerate "param regen". That should help you out.

meee1, 2021-01-06:
ive started to stage this already
https://firmware.ardupilot.org/Tools/MissionPlanner/MetaData/
however it wouldn't help in this exact scenario, as the firmware was a beta firmware.

Autonomost, 2021-01-06:
nice! Want any help with pointing mission planner to that link? Locking down what displays in the parameter lists?

meee1, 2021-01-06:
i created  the files.
but still need the method of ID'ing the firmware to correct xml name. and the appropriate loading etc.

Autonomost, 2021-01-06:
where in the code does MP grab the onboard firmware and display it at the top of the screen? This usually happens after all the parameters are pulled.

meee1, 2021-01-06:
https://github.com/ArduPilot/MissionPlanner/blob/master/MainV2.cs#L1754