ArduPilot/MissionPlanner #2503: bug in param download

tridge, 2020-11-12:
PARAM_VALUE data from a ftp download is corrupt for any fields that are not 32 bits wide. The problem seems to be that the constructed PARAM_VALUE is not initialising all bytes in the 32 bit float. It seems to be filling in some of the bytes as random data from the stack. This results in things like this:
```
2020-11-12 10:12:57.33: PARAM_VALUE {param_id : EK2_ENABLE, param_value : 1.401298464324817e-45, param_type : 2, param_count : 1158, param_index : 0}
```
the param_value has 3 garbage bytes in it, resulting in a nonsense value.
This needs to be fixed quickly as users downloading their parameters then uploading them again could cause arbitrary corruption of their parameters and potentially crash a vehicle

meee1, 2020-11-12:
the issue is around params packed via a union vs float.

meee1, 2020-11-17:
MP beta is not using floats for the mavftp param injection

closed 2020-11-17 by meee1