ArduPilot/MissionPlanner #2505: Servo/Relay tab to support RC_CHANNELS_OVERRIDE

magicrub, 2020-11-13:
Problem:
I'd like the ability to test servo ranges and exact PWM values on my aircraft. The Servo/Relay tab is a useful tool in testing and Q/A. If I set the SERVOn_FUNCTION to passthrough it works but we have to set the param back for normal operation, prone to human error. I'd like to do it without having to fuss with my params by sending an RC Input signal instead of a "servo" signal. Or maybe there's an easier way to do this that I don't know about?

Requests for Servo/Relay tab:
- Have checkbox to instead send mavlink msg RC_CHANNELS_OVERRIDE, allowing single or repeated msgs
- have all channels available
- add exact PWM text field (that is also bounded by the min/max)

For the repeated msg, I'm thinking of a "send" button (to replace toggle in this layout?) that changes to "stop" when pressed and keeps sending at 10Hz until clicked again. This would allow multiple streaming rc channels to be selectively enabled/disabled but you can't enable/disable them all at once. Another option would be to have "allow" checkboxes and a single start/stop stream button.

meee1, 2020-11-17:
"have all channels available"   top end or bottom end?

the intent is only for testing?  ie could this be just 1 row that has a channel/pwm selection/send option?

meee1, 2020-12-09:
@magicrub would a iron python script suit your purpose?   to send rc-override messages with one channel at a time?