ArduPilot/MissionPlanner #2506: Mission Planner fails to load. Throws outt an Exception  about an issue with 'log4net.version 2.0.8.0culture=neutral. Mission Planner is looking for a file but cannot find it.Hmm... any suggestions to fix this problem?

GROO254, 2020-11-15:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2020-11-17:
i would suggest reinstalling

GROO254, 2020-11-17:
Have just done so and the problem resolved itself. Seems that  the
"lastest" version of the software was installed before. Installing the
"stable" version solved the problem and the software is no longer fooking
for the log4net missing library. Thanks very much! Cheers!😀😀

closed 2020-12-01 by meee1