ArduPilot/MissionPlanner #2512: RTK/GPS Injection Position Accuracy & Obs Time not being set, uBlox Config in module takes precedence

PRIMETSS, 2020-11-27:
#### Issue details
Using a uBlox F9P module on the Ground Station MP.
The COM Port opens successfully and communicates with the uBlox GPS Module.
I set for example 2m & 30seconds for accuracy the "Survey In" shows a valid survey at 30m, which is set in the the uBlox's CFG/TMode 3 config.

If I set TMode to disabled, it appears to never get a valid SurveyIn
So it seems the autoconfig is not sending out the config properly?

I believe this is why out in field we get some accuracy issues, as the uBlox NVRam config was set for a very low accuracy just fo indoor diagnostics. But I didnt know the setting in MP RTK Config, was not over-writing that.
Just brought my Repo up-to-date as of today. Re-built and same results.

I'm trying to fault find it, but its slow progress. Question I have is, does the "RESTART" meant to re-burn the setting setting permanently into the uBlox? Or is it just a Hot Config push, so on reboot they are lost. As I seem to heard the USB port for the GPS drop, then reconnect after hitting Restart?

#### Version
1.3.74build 1.3.7636.29630

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Hardware type
PIXHawk 4, uBlox ZED-F9P GPS Module (FW version HPG 1.12

#### Logs
[image]
[image]

meee1, 2020-11-27:
the f9p remembers the last survey in location on reboots.
MP beta does resolve this when you click restart.

PRIMETSS, 2020-11-27:
Thankyou for that info appreciated!

I might be confused. But thought the F9P always did a re-survey on boot (using the Acc & Time settings in its NVRam config)
I understand if a Save a Location in MP, then that's what MP uses as current position (So I dont save location as base station is setup at different locations each deployment)

So the Acc & Time + Restart isn't currently working, is that just for the F9P and the M8P (which I was using previously)

If your wondering, the reason I'm trying to 'skip' this set, is I'm trying to make it as easy as possible for a non experienced operator to use. (also using touch tablet, which is difficult to hit buttons & type out in the field) So trying to have it re-survey each time its powered up. 

I can work around it by just burning in the field setting into the ublox. It's just that I'll forget and start chancing tail when go the gear on the bench inside and often want really low acc setting just to make sure RTK is roughly working on the bench before field trip.

meee1, 2020-12-01:
please try the current beta. i think i might have found an issue

PRIMETSS, 2020-12-01:
OK Thanks will do

PRIMETSS, 2020-12-01:
Thankyou! Appreciated! YES that seems to have resolved the issue. Now both Survey Accuracy & Time values are being used by the ublox F9P module when changed and hitting Restart and the limits are meet.
Also seem not to drop the USB comm port after hitting Restart, but will nee to try on laptop I noticed this on.

(ps: what was it?) Any idea when fix might hit the repo?

meee1, 2020-12-09:
its already in the repo, normally under the "beta" tag.   master sometimes is delayed however. as i sometimes rewrite the history on beta

closed 2020-12-09 by PRIMETSS