ArduPilot/MissionPlanner #2514: Mavlink: regenerate.bat not working

Belgacem-1, 2020-11-29:
#### Issue details
_After adding new mavlink message the regenerate.bat is not working_

#### Version
_master branch_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_
C:\WINDOWS\system32>set PATH=c:\python27;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Polyspace\R2019a\runtime\win64;C:\Program Files\Polyspace\R2019a\bin;C:\Program Files\Polyspace\R2019a\polyspace\bin;C:\Program Files (x86)\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\Microsoft SQL Server\110\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\PuTTY\;C:\Program Files\Git\cmd;C:\Users\RB\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\RB\AppData\Local\Programs\Python\Python39\;C:\Program Files\Java\jdk-13.0.1\bin;

C:\WINDOWS\system32>rd /s /q "mavlink"
Le fichier spécifié est introuvable.

C:\WINDOWS\system32>rem python -m pymavlink.tools.mavgen --lang=C --wire-protocol=2.0 "message_definitions\ardupilotmega.xml"

C:\WINDOWS\system32>python -m pymavlink.tools.mavgen --lang=CS --wire-protocol=2.0 "message_definitions\ardupilotmega.xml"
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "...\mavlink\pymavlink\tools\mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "...\mavlink\pymavlink\generator\mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future

C:\WINDOWS\system32>rem python -m pymavlink.tools.mavgen --lang=CS_OLD --wire-protocol=2.0 "message_definitions\ardupilotmega.xml"

C:\WINDOWS\system32>copy /y "mavlink\mavlink.cs" "Mavlink.cs"
Le chemin d’accès spécifié est introuvable.

C:\WINDOWS\system32>pause
Appuyez sur une touche pour continuer..._

meee1, 2020-12-01:
you need to install "future" into your python 2.7 install.

"ImportError: No module named future"

Belgacem-1, 2020-12-01:
Thank you for return @meee1 I did it before posting this issue it seems due to python version, for mine i have 3.9.0, i uninstalled it and reinstalled 2.7.18 and future, the problem solved but this ( C:\WINDOWS\system32>rd /s /q "mavlink"
Le fichier spécifié est introuvable.) remain appeared in terminal. Didn't find the mavlink directory to remove it

meee1, 2020-12-01:
dont run it as admin, the local path should be inside the directory, for some reason you are in the windows system directory

Belgacem-1, 2020-12-01:
Yes you are right, thank you so much @meee1

closed 2020-12-01 by Belgacem-1