ArduPilot/MissionPlanner #2516: Speech synthizer language mismatch

Ilia-SB, 2020-12-08:
#### Issue details
Since long ago I had a weird problem with speech in MissionPlanner: I have UI language set to "English (US)" but my system's (Win10) language is Russian. As a result all the speech in MP sounds as english text read by a non-english speaker which is very annoying. Looks like MP uses the default system language for Text To Speech which results in problems when synthizing text in a different language.
I actually managed to confirm this by a simple code modification: in Speech.cs I added the following 2 lines to **SpeakAsync(string text)**:
```
    PromptBuilder pb = new PromptBuilder(new System.Globalization.CultureInfo("en-US"));
    pb.AppendText(text);
```
and changed
`    _speechwindows.SpeakAsync(text);`
to
`    _speechwindows.SpeakAsync(pb);`

Now speech works just fine.
I guess PromptBuilder should be initialized with the culture corresponding to the interface language set in MP. Unfortunately, I have very limited experience with Github, and more importantly no knowledge of MP code. So I won't volunteer to fix it myself. Shouldn't be hard for someone knowledgeable though.

#### Version
1.3.74 and below

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
All

#### Hardware type
pixhawk

#### Logs
None

meee1, 2020-12-09:
Thanks please try the current beta MP

Ilia-SB, 2020-12-09:
Yep, works as expected.
Thanks!