ArduPilot/MissionPlanner #2519: "do this for all" when loading params

tridge, 2020-12-16:
When loading a full param file you can quite often get a lot of "marked readonly" and "out of range" dialog boxes. It should be nice to have a "do this for all" checkbox in the dialog to make this is a bit faster

roberthudson, 2021-03-03:
I took a fresh look at this issue...

I just don't like the current approach we're on. Playing "Whack-a-Mole" with dialog boxes doesn't really make sense.
Imagine this, if in Visual Studio there's a missing reference there might be 100+ compile errors. Do I want to see 100 Messageboxes? Heck no! I don't want to see even **one** Messagebox.
I want to see a **_list_** of errors.

If it were up to me, I'd forget about the "Do for all" checkbox

Here's how I would modify MissionPlanner:
-Click Load Params
-The params all load into an array, like they are now.
-If there are one or more issues, display a new grid

In the grid the user can see all params which are "read only" or "out of range"
They could then Select All/Select None and apply the values or disregard the values. Or rollback the entire load

Besides, the current dialog box is ambiguous. "Do you want to continue?"
Continue what? Continue applying the quirky value? Or continue loading the params?

roberthudson, 2021-03-03:
Something like ....

[image]

tridge, 2021-03-03:
> I took a fresh look at this issue...

I like this idea a lot. @meee1  ?

meee1, 2021-03-03:
yes agreed, much cleaner, and seems alot more fit for purpose.

roberthudson, 2021-03-06:
Ok, I'm getting to know Mission Planner more and more.
So..... I'm rethinking this issue yet again. We're still on the wrong approach.

Preface at @meee1: No disrespect in the following. Mission Planner is just amazing and you've done an incredible job with it. But with system this complex, sometimes small details get overlooked.
I've been asked to look at one particular issue (the Whack-a Mole dialog game) - so that's where I'm focusing.

There's a bit of a logic flaw. Here's why we're getting all those dialog boxes. They popup on the CellValueChanged event.
This event fires for each row when the grid is loaded. What we need is a bool flag "IsLoading" that is set to true when the grid is populated. Then clear the flag.
While IsLoading=true we want to ignore if the value is ReadOnly. Who cares? We only care IF the user tries to CHANGE it.

Here's what I suggest:
-set an IsLoading flag and suppress the MessageBox while the grid is populating. Then clear the flag.

-then, if the user tries to change a ReadOnly cell we show the dialog. Lock the cell so they can't change it. Mark the cell in orange or something.

-we can scrap the Do This For All checkbox. It's not needed

-for the Out Of Range cells, mark them in some other color as a visual cue. Maybe yellow

-after populating the grid display a dialog "n values are out of range. They have been highlighted yellow"

(Sorry for the repost, this should be here not in the PR comments)

tridge, 2021-03-08:
> -then, if the user tries to change a ReadOnly cell we show the dialog. Lock the cell so they can't change it. Mark the cell in orange or something.

the ReadOnly issue almost always comes up when users load a set of parameters from a file.
What happens is users share parameter files with each other, or use param files from previous flights and want to load those files on the vehicle. The param files are typically around 1000 parameters.
When they load those param files there are often quite a few parameters that are readonly or are out of range. That is what generates the warnings