ArduPilot/MissionPlanner #2522: [feature request] message button

IamPete1, 2020-12-20:
It is possible to send a message to AP that it then records in its log, this uses the GSC send test message. The AP receiving implantation is here:
https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L2833

In MavProxy this is possible with the `msg` module for example:
```
module load msg 
msg test of message functionality
```

[image]

The use case is if you notice something in the flight, or want to note a particular time you can send the note and it will be induced in the AP dataflash log.

magicrub, 2020-12-21:
Great idea!

IamPete1, 2022-04-20:
closed by https://github.com/ArduPilot/MissionPlanner/pull/2841

closed 2022-04-20 by IamPete1