ArduPilot/MissionPlanner #2523: Beta Downloads

auturgy, 2020-12-21:
#### Issue details
In firmware.ardupilot.org "latest" downloads for firmwares are latest master.  For MissionPlanner, "latest" is latest stable.  This is confusing - it would be beneficial to standardise what "latest" means across the project.  There is also no convenient zip download for latest MissionPlanner beta - which is problematic for linux users, as updates don't stick unless running under sudo.

#### Version
latest

#### Platform
firmware.ardupilot.org

meee1, 2020-12-21:
i think changing this now will cause a large number of dead links

meee1, 2020-12-21:
ive added a stable symlink, but its just a copy of latest

meee1, 2020-12-21:
all beta/master builds of MP come from CI

meee1, 2020-12-21:
if you want a url to the zip file MP uses

https://ci.appveyor.com/api/projects/meee1/missionplanner/artifacts/MissionPlannerBeta.zip?tag=beta