ArduPilot/MissionPlanner #2526: ARM Linux build?

Botspot, 2020-12-24:
It would be awesome if I could run MissionPlanner on a Raspberry Pi desktop computer.

meee1, 2020-12-25:
this is already being done. check the forum
https://discuss.ardupilot.org/t/how-to-guide-mission-planner-on-raspberry-pi/60889

closed 2020-12-25 by meee1

Botspot, 2021-01-01:
Okay it worked fine after I disabled the OpenGL driver.  
I didn't want want to disable the OpenGL driver system-wide, so I found an alternative:
```
LIBGL_ALWAYS_SOFTWARE=1 mono /home/pi/MissionPlanner-latest/MissionPlanner.exe
```
Now it works great, and all my other software still has access to the OpenGL driver! :)

@meee1 Could you forward this information to that discuss.ardupilot.org thread please, and maybe get the pdf instructions updated?