ArduPilot/MissionPlanner #2527: Elevation data from the Norwegian Mapping Authority.

Supermagnum, 2020-12-27:
#### Issue details
Please implement the free and open DTM10 data from Kartverket aka the Norwegian Mapping Authority. It’s in 10x10 meter grid measurements, with a accuracy of ± 4 to 6 meters,as it’s superior to any SRTM data.

Preprocessed national dataset exports are available.
All datasets are in EUREF89, NN2000. UTM zone 33 has full coverage. Zone 32 and 35 where applicable. It is available here:
https://kartkatalog.geonorge.no/metadata/height-dtm-10/dddbb667-1303-4ac5-8640-7ec04c0e3918
There is also a API.
https://kartkatalog.geonorge.no/metadata/height-dtm-50/e25d0104-0858-4d06-bba8-d154514c11d2

#### Platform
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

rmackay9, 2020-12-27:
I suspect this is similar to this issue we found for finland https://github.com/ArduPilot/ardupilot/issues/12923

Supermagnum, 2020-12-28:
> I suspect this is similar to this issue we found for finland [ArduPilot/ardupilot#12923](https://github.com/ArduPilot/ardupilot/issues/12923)

It looks like it is. I have downloaded the entire dtm data set today, it was around 6 GB .

meee1, 2020-12-28:
if you have the data as a geotiff, you can use it inside MP right now. https://discuss.ardupilot.org/t/custom-imagery-in-mission-planner/10280/13?u=michael_oborne

Supermagnum, 2020-12-28:
This looks to be 1 meters resolution:
https://kartkatalog.geonorge.no/metadata/hoeyde-dom-1-utm32-wcs/08edfa0c-0e8f-44eb-80a6-ddbf3d1b3a64

Supermagnum, 2021-01-13:
All data is licensed under Creative Commons BY 4.0 (CC BY 4.0).

Supermagnum, 2021-01-13:
> I suspect this is similar to this issue we found for finland [ArduPilot/ardupilot#12923](https://github.com/ArduPilot/ardupilot/issues/12923)

It looks to be the same. 
The DTM sets that is available as WMS services are the top 6 links here:
https://kartkatalog.geonorge.no/metadata/hoeyde-dtm-1-utm35-wms/9ecc47f9-2d10-4387-b033-90d99e104980

https://kartkatalog.geonorge.no/metadata/hoeyde-dtm-1-utm33-wms/4d07b725-40c1-455a-a3b3-7398bb4022d1

https://kartkatalog.geonorge.no/metadata/hoeyde-dtm-1-utm32-wms/c86c9db6-90d5-4b23-9b15-673050b35d33

https://kartkatalog.geonorge.no/metadata/hoeyde-dtm-1-utm35-wcs/bac437e4-748d-4748-b2b0-975a1a694cf0

https://kartkatalog.geonorge.no/metadata/hoeyde-dtm-1-utm33-wcs/3776e807-8ae6-4a9c-8481-e2c9f6333e23

https://kartkatalog.geonorge.no/metadata/hoeyde-dtm-1-utm32-wcs/600e3b86-64bd-4c31-8b26-a552a621c5a8

Together it looks like they completes the coverage of Norway.

Supermagnum, 2021-01-26:
> 
> 
> if you have the data as a geotiff, you can use it inside MP right now. https://discuss.ardupilot.org/t/custom-imagery-in-mission-planner/10280/13?u=michael_oborne

The files are in :

EUREF89 UTM sone 33, 2d
TIFF 

The files are in the correct folder, but I am unsure if the format is correct.
https://www.dropbox.com/sh/3szyzil8j7hvcuj/AACGQB5EYQ1BgmkVJbPqLwxSa?dl=0

Can anyone look at those ?

Supermagnum, 2021-03-10:
DEM files are available, but does Mission planner or APM planner support those ?

mbheir, 2021-03-25:
@Supermagnum I'm having the same issue. Is there a way to use the DTM10/30 datasets from Norwegian MA in Ardupilot? (without running everything trough MP)

meee1, 2021-03-25:
https://kartkatalog.geonorge.no/metadata/dtm-10-terrengmodell-utm33/dddbb667-1303-4ac5-8640-7ec04c0e3918
you can grab the geotiffs from this link. im looking at supporting ETRS89, but in the mean time just need to convert them to WGS84 and they will work as is

meee1, 2021-03-25:
after a quick read it seems they are similar, apart from one deals with tectonic movement, one does not.

meee1, 2021-03-25:
ive just added support to beta MP

mbheir, 2021-03-26:
I tried out the beta, using a DTM10 TIF from the link your comment and it loaded in MP! However, there are some major issues. 
I put a few screenshots to illustrate. The yellow rectangle shows where MP says I have Geotiff data, and the red shows where I'm supposed to have Geotiff data (very ish). (you can verify this by looking at the black/white elev map too. could it be rotated). Also the elevation data is very strange, and looks misplaced. 

[image]
[image]

meee1, 2021-03-26:
can you try the new beta again. just push a fix

meee1, 2021-03-26:
tip, use Control-W and turn on terrain

meee1, 2021-03-26:
i just found another bug, so hold off

meee1, 2021-03-26:
[image]
example

Supermagnum, 2021-03-26:
> after a quick read it seems they are similar, apart from one deals with tectonic movement, one does not.

Not that I am aware of.  We just have to wait for this request to be resolved. I am hoping for 1 meters resolution.

Supermagnum, 2021-03-26:
Note: 
The last files for DTM10 that were updated are from 2013. The Norwegian Mapping Authority instead prioritises the establishment of the National Detailed Elevation Model (hoydedata.no), where new elevation models are made available as they are produced. At hoydedata.no there are also ready-made exporters of nationwide terrain models with a resolution of 10 meters. These are generated from detailed laser data where available. This is supplemented with height data from the 2013 edition of DTM10 if necessary to get full coverage. 1 meters resolution is also available, it appears to be constantly updated.

mbheir, 2021-03-26:
The DTM10 dataset now works like a charm! It also loads on the APM as expected. Thank you so much, this is so much better than the srmt data!

[image]

Supermagnum, 2021-04-10:
> The DTM10 dataset now works like a charm! It also loads on the APM as expected. Thank you so much, this is so much better than the srmt data!
> 
> [image]

How did you get the geotiff files to load in Apm planner?
There are no documentation that mentions where to put such files.

rmackay9, 2024-04-24:
I think we can close this issue now but please tell me if I've got it wrong.  We also have SRTM for Norway now too.

closed 2024-04-24 by rmackay9