ArduPilot/MissionPlanner #2528: DO_SET_ROI invalid parameters

tridge, 2020-12-27:
@Hwurzburg is getting this FPE for a DO_SET_ROI from MissionPlanner in SITL. param5 and param6 look like invalid values

[image]

[image]

Hwurzburg, 2020-12-27:
@meee1 Michael, fyi, I was testing point camera here on a new RF model with gimbal...if I use the MP SITL, it seems to work, but if I am using the AP SITL and forward from Mavproxy to MP from the Linux box on the WIndows box running RF, then this happens on any point camera here comand...

meee1, 2020-12-27:
@Hwurzburg can i get some more info on how in MP you are sending the command?