ArduPilot/MissionPlanner #2530: GPS track appears to jump from GPS[0] to GPS[1] in logs with two GPS receivers

arikrupnik, 2021-01-04:
#### Issue details
When displaying logs with two GPS receivers, the graphs seem to jump from primary to secondary GPS. When the GPSs' readings differ, this creates a zigzag pattern in the track.

[image]

#### Version
1.3.74

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Fixed-wing, though probably airframe-independent

#### Hardware type
Beaglebone Blue, u-blox M8, VectorNav VN-300

#### Logs
https://github.com/arikrupnik/ardupilot-config/blob/master/logs/FIXED_WING/7/2021-01-02%2017-09-42.bin?raw=true

tridge, 2021-01-04:
I suspect this is related to the change from GPS and GPS2 to using GPS[0] and GPS[1]

rmackay9, 2021-01-04:
I think this is fixed in the latest beta.

meee1, 2021-01-04:
yes, this was fixed in beta about a month ago

closed 2021-01-04 by tridge