ArduPilot/MissionPlanner #2533: Cant find code to receive message by  another System id

lycn2020, 2021-01-06:
#### Issue details
On the source code of Mission Planner I want to receive messages from mavlinke by system ID like 244 
What is the code in C# to receive message by System ID
Please Help me

#### Version
1.3.74
#### Platform
source code of Mission Planner

#### Airframe type

no
#### Hardware type

no
#### Logs
no
[image]

meee1, 2021-01-06:
if its a valid mavlink packet, it will show up already

lycn2020, 2021-01-06:
i sent  a packet from arduino with sys id 244 . and i wanna show it on mission planner

meee1, 2021-01-06:
if mission planner not displaying it, then the packet is invalid

lycn2020, 2021-01-06:
[image]
 it displayed in mission planner (mavlink inspector).but how to use it in another form.

closed 2021-01-17 by lycn2020