ArduPilot/MissionPlanner #2535: flightaxis SITL - incorrect current reading when negative values

nhadrian, 2021-01-08:
#### Issue details
When I'm using SITL with RF9 and flighaxis controller, negative current values in RF9 (when motor is braking) gets false high readings in Mission Planner. This causes false drop in battery percentage. It seems like an issue with signs and flightaxis/MP link.

RF9 current reading (-1.1A):
[image]

Mission Planner current reading (327,7A): 
[image]

#### Version
1.3.74.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
plane, quadplane

#### Hardware type
SITL / flightaxis

meee1, 2021-01-08:
do you have a tlog of this as well?

nhadrian, 2021-01-08:
Yes, in this short tlog after takeoff (at about 2/3 time of log length) you can see multiple times when current in RealFlight changes to negative, MP shows 327,7A.
[2021-01-08 09-11-30.zip](https://github.com/ArduPilot/MissionPlanner/files/5785993/2021-01-08.09-11-30.zip)

meee1, 2021-01-08:
ok so this looks like an ardupilot bug
the field over mavlink is signed, so to get that value is bad
if you graph current and battery remaining, there is a large jump in capacity
[image]

i would suggest posting the issue in the arudpilot issue list, and link to this issue

IamPete1, 2021-01-16:
I'm fairly sure that the Realflight NavGuides are just nonsense and that there is no underlying bug in AP. This can probably be closed.

https://github.com/ArduPilot/ardupilot/issues/16272#issuecomment-761693444