ArduPilot/MissionPlanner #2538: Open Street Map (OSM) don't work

joaomarcelocorrea, 2021-01-12:
When I choose the Open Drone Map it only displays this message: "We are sorry, but we don't have imagery at this zoom level for this region"
I'm from Brazil. Version 1.3.70 works.

meee1, 2021-01-13:
fixing this in beta MP. seems OSM was blocking mission planner

Faiyaz349, 2021-02-09:
> When I choose the Open Drone Map it only displays this message: "We are sorry, but we don't have imagery at this zoom level for this region". I'm from Brazil
> Mission Planner version 1.3.74.1
> In the version 1.3.70 OSM works.

Hello. Have you fixed this issue? 
I am getting the same problem in MIssion Planner 1.3.70 and even in old versions also. 
Can you please update me if you found the result.
Thanks In advance. 
[image]

closed 2025-04-07 by meee1