ArduPilot/MissionPlanner #2542: Writing WP Mission to AC 3.5.5 Fails

itayguy, 2021-01-14:
#### Issue details
Writing WP Mission to AC 3.5.5 Fail using Mission Planner 1.3.74
#### Version
1.3.74
#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter v 3.5.5
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
quad

#### Hardware type
pixhawk

#### Logs

[image]

itayguy, 2021-01-15:
No , it works fine with 4.x releases

itayguy, 2021-01-23:
debugging the issue , seems the msg is 12 bytes vs 16 bytes in the newer version , missing target_system and target_component
[image]
[image]

gozeberke, 2021-05-23:
I have a same problem. How can i solve this problem?