ArduPilot/MissionPlanner #2547: Enhancement suggestion: edit waypoint lat/long by entering distance and azimuth

DaveE128, 2021-01-25:
#### Issue details
_Please describe the problem, or desired feature_
Add or edit waypoints by entering the distance and azimuth from the previous waypoint. When planning missions, it would be very useful to be able to enter the distance and bearing from the last waypoint and have mission planner calculate the lat/long automatically. It would sometimes also be useful to enter the gradient/angle and have the altitude calculated automatically.

I am having to do this in a spreadsheet myself and transferring into mission planner is a hassle.
#### Version
_What version was the issue encountered with_
1.3.74
#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
n/a
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
n/a
#### Logs
_Please provide a link to any relevant logs that show the issue_
n/a

meee1, 2021-01-27:
so you  want COGO input?

DaveE128, 2021-01-27:
I guess that's the same thing!

At the moment I'm using forumlae in excel much like those here:
https://stackoverflow.com/questions/7222382/get-lat-long-given-current-point-distance-and-bearing#:~:text=asin%20%3D%20arc%20sin()%20d,radians%2C%20clockwise%20from%20north)%3B

or here:
http://www.movable-type.co.uk/scripts/latlong.html

I was thinking interface-wise, you could just edit the "Dist" and "AZ" columns in the plan view and the lat/long for that waypoint would be recalculated to match.

Thanks!