ArduPilot/MissionPlanner #2555: MP Turn Dia (at 45d) as displayed in Flight Planner Incorrect ?

jagrandle, 2021-02-04:
#### Issue details
It looks like the calculation for the Turn Diameter at a 45deg Bank may be incorrect. This number is very useful for determining whether alternating lanes etc need to be used for a survey mission. It is only useful if it correct.

Standard equations for this are given at the bottom of this web page.
https://en.wikipedia.org/wiki/Standard_rate_turn

A couple of examples are also given here, where a speed of 50m/s gives a turn radius at 45deg bank of 255m (diam = 510m) yet MP gives a Turn Dia (at45d) of 728m ?
https://aviation.stackexchange.com/questions/2871/how-to-calculate-angular-velocity-and-radius-of-a-turn

Thank you

#### Version
1.3.74

#### Platform
n/a

#### Airframe type
n/a

#### Hardware type
n/a

#### Logs
n/a

meee1, 2021-02-04:
which screen is incorrect? ie flight planner does not have a turn radius box

if its the survey grid screen you are correct. i have 35 in code, and 45 in the description

meee1, 2021-02-04:
https://github.com/ArduPilot/MissionPlanner/commit/7f5a27b33540c64d15a2eafaca7a227b1a76629b

jagrandle, 2021-02-04:
Hi, sorry for the confusing terminology. I wasn't quite sure of the screen name, but it is Survey (Grid) as you correctly surmised.

From my point of view if you could change the calculation to 45deg to match the description that would be great. RPAS typically seem to have higher performance than manned aircraft and I certainly have a number of RPAS set up to achieve at least a 45deg banked turn.

Thanks for looking at it so promptly.
Cheers