ArduPilot/MissionPlanner #2557: Python Scripting add-on

TheOracle2, 2021-02-11:
Let me explain...
I'm a proficient Python programmer. I enjoy creating libraries but a couple of years ago, me and my dad dabbled in creating a drone. I proudly now am able to declare myself a pro at Ardupilot after several crashes. Fast forward to last night when I was brushing my teeth and recalling how hard it was to create a optional advancements for my V-TOL. So I propose that I and someone else can work together to add a Python library with a scripting console to Mission planner for people to create scripts (obviously!). I need someone who can C# and a little bit of python but has the know-how to implement this idea. I think this is an idea that can increase the speed of a prototyping workflow and add more functionality to Ardupilot, a already great idea. 

I know a similar version of this idea has already been implemented, but I also want to add a couple of features as listed below!
This could be done via functions/methods if any technical introduction is needed.

- A PWM tester
- Communication in-air
- Linkable python scripts to actions such as switches...
- Waypoints but with distances
- AI (Machine Learning??)
- Sharable scripts
- Quick prototyping
- Testing new hardware add-ons (Like creating a ISquareC tool and testing it)
- Serial output/input
- etc. as suggested below...

Any support is welcomed and God Bless!
Thank You!
tl;dr : Yes I know my spiel is long, but I want to create a python scripting system to create a tool to add new functionalities through Ardupilot.

meee1, 2021-02-12:
this already exists.
Mission planner has a iron python implementation already working. and a few people are using this already. i even ship examples in the "scripts" folder

TheOracle2, 2021-02-13:
Okie. Can we expand on that?? I do like the current functionality but I believe that it isn't the best it could be. Not tryna be pushy
but are all the features above implemented?

liguang426, 2023-04-25:
在mission planner脚本系统方面我遇到点问题，我们可以讨论一下吗？在脚本编写中我调取pymavlink库，在运行脚本是显示并没有这个模块，我很苦恼