ArduPilot/MissionPlanner #2563: GeoTagging New Feature: Add Roll, Pitch, Yaw to Exif data

jmachuca77, 2021-02-22:
Pix4Dmatic needs pitch, roll and yaw data to be embedded in the exif data for it to be able to work with the pictures.
Information on what is needed is here:
https://support.pix4d.com/hc/en-us/articles/205732309-EXIF-and-XMP-tag-information-read-by-Pix4D
As far as I can tell we already have the pitch, roll and yaw logged into the CSV file so its just a matter of adding it to the exif data just not sure how to do that.
[image]

meee1, 2021-02-23:
XMP must be new, but it has been a while since i looked at this stuff

meee1, 2021-02-23:
https://support.pix4d.com/hc/en-us/articles/360016450032-Specifications-of-xmpcamera-tags

jmachuca77, 2021-02-23:
Here is an image with the data embedded
https://www.dropbox.com/s/mizx6cetmgw9zjq/ix-02-81060_0024_0906.jpg?dl=0

cyhly, 2021-03-02:
when i using the map baering,the marker is missing.
when i set the mar bearing equal to zero,the marker is visible.
what shoule i do?