ArduPilot/MissionPlanner #2573: Error while building mission planner

Jevanraj, 2021-03-10:
#### Issue details####
Hello, when I tried to build mission planner in Windows 10, I have 3 errors:

Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\user\source\repos\Filetest.razor\ExtLibs\7zip\Compress\LzmaAlone\obj\Release\.NETFramework,Version=v4.0.AssemblyAttributes.cs	4	Active
Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\user\source\repos\Filetest.razor\ExtLibs\7zip\obj\Debug\netstandard2.0\.NETStandard,Version=v2.0.AssemblyAttributes.cs	4	Active
Error		Could not find file 'C:\Users\user\source\repos\Filetest.razor\ExtLibs\7zip\bin\Debug\net461\7zip.dll'.	MissionPlanner	C:\Users\user\source\repos\Filetest.razor\test\FirmwareSelection.xaml

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub
[image]

custeve, 2021-03-23:
Ditto.

main2stels, 2021-03-25:
The same problem

meee1, 2021-03-26:
update your VS version.
its currently building in CI ok, so its something with your setup

bobf12, 2021-04-21:
Hi. 
I have what looks like a similar problem. I'm sure it is something to do with my setup - but any help about where the problem might be would be much appreciated.
I'm using Visual Studio 2019 16.9.4.

When I open and build (using Build Batch - as in the instructions) the MissionPlanner solution I get lots of errors (often around 48 - but the number does seem to vary!), many  to do with 7zip.dll or MissionPlanner.exe not being found. A sample below.
I've added the path to the installed MissionPlanner to the properties for the overall MP solution - but can't see how to do so for the other projects. 
Also 

Any ideas?
Many thanks!

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\User\source\repos\MissionPlanner\ExtLibs\7zip\Compress\LzmaAlone\obj\Release\.NETFramework,Version=v4.0.AssemblyAttributes.cs	4	Active

Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\User\source\repos\MissionPlanner\ExtLibs\7zip\obj\Debug\net461\.NETFramework,Version=v4.6.1.AssemblyAttributes.cs	4	Active

Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\User\source\repos\MissionPlanner\ExtLibs\7zip\obj\Debug\netstandard2.0\.NETStandard,Version=v2.0.AssemblyAttributes.cs	4	Active

Error	CS0006	Metadata file 'C:\Users\User\source\repos\MissionPlanner\ExtLibs\7zip\bin\Debug\net461\7zip.dll' could not be found	MavlinkMessagePlugin	C:\Users\User\source\repos\MissionPlanner\ExtLibs\MavlinkMessagePlugin\CSC	1	Active

meee1, 2021-04-21:
try building just Mission Planner. what happens

bobf12, 2021-04-21:
Thanks for the speedy reply!

Just tried building only the MP (from the menu in the Solution Explorer). Only 4 errors now (below), but pretty much the same thing going on: 
I'm guessing (you can tell that I'm not a VS user or Windows developer!) that duplicate versions of something are preventing 7zip in ExtLibs from building, that in turn causes the whole build to fail.

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\User\source\repos\MissionPlanner\ExtLibs\7zip\Compress\LzmaAlone\obj\Release\.NETFramework,Version=v4.0.AssemblyAttributes.cs	4	Active

Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\User\source\repos\MissionPlanner\ExtLibs\7zip\obj\Debug\net461\.NETFramework,Version=v4.6.1.AssemblyAttributes.cs	4	Active

Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\User\source\repos\MissionPlanner\ExtLibs\7zip\obj\Debug\netstandard2.0\.NETStandard,Version=v2.0.AssemblyAttributes.cs	4	Active

Error		Could not find file 'C:\Users\User\source\repos\MissionPlanner\ExtLibs\7zip\bin\Debug\net461\7zip.dll'.	MissionPlanner	C:\Users\User\source\repos\MissionPlanner\test\FirmwareSelection.xaml

meee1, 2021-04-21:
do you have these installed
[image]

bobf12, 2021-04-22:
I was missing a couple of those things (the C++ one and VS extension development ones that I guessed weren't necessary). I've now installed them - and the result is the same.
I'll keep trying things - but am running out of ideas!

ohitstarik, 2021-04-29:
Did anybody find a solution to this? Having the same problems right now with 7zip stuff, even after updating and adding .dlls.

bobf12, 2021-04-29:
I got around the problem in the end - but more by admitting defeat than doing anything clever. 
I re-installed Visual Studio (I had been using the latest version, but it much have got messed up somehow), and worked on a new clone of Mission Planner.
After that, doing a batch build of everything reduced the number of errors from loads down to 2 (copied below, in case anybody's interested). One was to do with the wasm project, and the other to do with a reserved file name being used. I couldn't find either discussed online - quite baffling.

Happily, building just the MissionPlanner seems to work fine. I can now change things and rebuild, and it seems to work!

Good luck getting your build working.

Error	APPX0701	Reserved file name 'AppxManifest.xml' is used by payload item 'C:\Users\User\Documents\GitHub\MissionPlanner\appxmanifest.xml'.	WindowsStore	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VisualStudio\v16.0\AppxPackage\Microsoft.AppXPackage.Targets	2959	

Error	NETSDK1004	Assets file 'C:\Users\User\Documents\GitHub\MissionPlanner\ExtLibs\wasm\obj\project.assets.json' not found. Run a NuGet package restore to generate this file.	wasm	C:\Program Files\dotnet\sdk\5.0.202\Sdks\Microsoft.NET.Sdk\targets\Microsoft.PackageDependencyResolution.targets	241

ohitstarik, 2021-04-29:
just did the same thing - but i cloned the git repository through terminal now and made sure to update submodules. everythings compiling just fine. Thanks for the help!

Kyros5, 2024-01-02:
I have this problem Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	CS0579	Duplicate 'global::System.Runtime.Versioning.TargetFrameworkAttribute' attribute	7zip (netstandard2.0)	C:\Users\LENOVO\source\repos\MissionPlanner\ExtLibs\7zip\Compress\LzmaAlone\obj\Release\.NETFramework,Version=v4.0.AssemblyAttributes.cs	4	Active	
any help in this issus

Kyros5, 2024-01-02:
Hello Guys I need help for build the mission Planner i try to fix the error, if any one know the solution