ArduPilot/MissionPlanner #2578: firmware load fails to PX4 mini with "unrecognized token at index 1..." error

mattdholmes00, 2021-03-24:
#### Issue details
Cannot load new firmware onto Pixhawk4 Mini.  Have tried multiple computers, different firmware versions.  I have checked that I have the appropriate com port selected (though remaining "unconnected").  

I receive this error:

"Error loading firmware file
System.Exception: Unrecognized token at index1
at fastJSON... (there's more; full text included in screengrab)" 

[image]

#### Version
Using MissionPlanner 1.3.74 and trying to load 4.1.0 obtained via:

https://firmware.ardupilot.org/
> copter
> latest
> PH4-mini
> arducopter_with_bl.hex

I'm at a loss as to how to proceed.

meee1, 2021-03-25:
Missionplanner only supports the .apj's

mattdholmes00, 2021-03-25:
Oh, ok, that was not clear at all.  thanks and sorry for bothering everybody

closed 2021-03-25 by mattdholmes00